ros-simulation/gazebo_ros_pkgs #513: gazebo_ros_control is failing to build on wily and jessie but ok on xenial

tfoote, 2016-11-29:
http://repositories.ros.org/status_page/ros_kinetic_dj.html?q=REGRESSION
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=REGRESSION

It looks like this is a regression recently and I see that ros_control was released recently. https://github.com/ros/rosdistro/pull/13320 by @bmagyar

bmagyar, 2016-11-29:
Weird...The logs suggest it's related to the unboosting efforts over at urdf.
Related PR on `ros_control`: https://github.com/ros-controls/ros_control/pull/251

It seems to me that the most relevant error is this:
```
/opt/ros/kinetic/include/joint_limits_interface/joint_limits_urdf.h:51:28: error: ‘JointConstSharedPtr’ is not a member of ‘urdf’
```

How do other dependants of urdf do?

tfoote, 2016-11-30:
@wjwwood what was the approach you recommended from testing with rviz?

wjwwood, 2016-11-30:
The urdf package has a comparability header now.

bmagyar, 2016-11-30:
Is there any action I need to take? Please point me at something :)

tfoote, 2016-12-06:
@bmagyar I think that the join_limits_interface needs to include this newly added header in urdf: https://github.com/ros/robot_model/commit/f5a3b8048c9fb4767ebd20b0cd80531c81522017 Which should give you the pointer referenced on all platforms. It looks like the same issue is breaking the schunk_canopen_driver: http://build.ros.org/view/Kbin_uW64/job/Kbin_uW64__schunk_canopen_driver__ubuntu_wily_amd64__binary/28/console

bmagyar, 2016-12-07:
Thanks! I've submitted a patch, hoping to get some results from jenkins soon, then I'l do the release.

tfoote, 2017-12-21:
This was resolved a while ago.

closed 2017-12-21 by tfoote