ros-simulation/gazebo_ros_pkgs #529: block laser plugin inverting heights

kstolleis, 2017-01-11:
Using the current version of block laser plugin results in z-axis heights being inverted in point cloud and confirmed in RViz. In March 2014 a commit was made 8440a0c5c9ac8078fb4b0bdfc1eebff495340c1e which made changes to the plugin where the z-axis height was negated.

A fix was proposed on Jade branch to remove the negation on lines 336 and 346 but was never merged into master.  I made this change and rebuilt from source and testing shows fix is correct.  Not sure why it was not accepted but should be tested and merged asap.

Issue with a single beam is not obvious but using a wide vertical field of view will make the problem quite apparent.

j-rivero, 2017-05-17:
Thanks for the warning and confirming the bug. Could you please create a pull request addressed to kinetic-devel and link this issue from there?

ompugao, 2017-05-18:
I created a pull request. #582

ompugao, 2017-05-18:
> A fix was proposed on Jade branch to remove the negation on lines 336 and 346 but was never merged into master.

i see, #458 is too big to merge.

j-rivero, 2017-05-18:
Merged, thanks. It will be ready in the next version of the packages, already available from source.

closed 2017-05-18 by j-rivero