ros-simulation/gazebo_ros_pkgs #531: Can't setup GAZEBO_PLUGIN_PATH with package.xml

abaudot, 2017-01-13:
Hello,

My package.xml contains the following `<export>` tags :

    <export>
        <gazebo_ros gazebo_model_path="${prefix}/models"/>
        <gazebo_ros gazebo_plugin_path="${prefix}/plugins"/>
    </export>

But it appears that when I run a .launch file from the package, only GAZEBO_MODEL_PATH is set, and not GAZEBO_PLUGIN_PATH. Thus, my models load, but not my plugin.

What am I doing wrong ?

(More detail on what I am trying to achieve [here](http://answers.gazebosim.org/question/15200/cant-load-plugin-with-roslaunch/), but it boils down to this current issue)

Cheers

vincentrou, 2017-02-16:
Did you try to put them on the same gazebo_ros tag ?

abaudot, 2017-02-16:
Thank you for your answer.

Not sure what I tried at the time, but now I did. Still no plugin is found.

ironmig, 2018-06-15:
Looking at [the code](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros/src/gazebo_ros_paths_plugin.cpp#L75), the tag that is recognized by the path plugin is actually "plugin_path" not "gazebo_plugin_path". Try:
```<gazebo_ros plugin_path="${prefix}/plugins"/>``` instead. This is inconsistent with the other tags and should probably be changed... did you find any documentation that led you to believe it is gazebo_plugin_path?

ironmig, 2018-06-15:
As this is just a usage issue, closing.

closed 2018-06-15 by ironmig