ros-simulation/gazebo_ros_pkgs #536: Gazebo 8 support

BenBlumer, 2017-02-08:
I've [installed](http://answers.gazebosim.org/question/15141/is-it-possible-to-use-ros-gazebo_ros-with-the-recent-nightly-builds-of-gazebo8/?answer=15398#post-id-15398) Gazebo 8 with ROS Kinetic on Ubuntu 16.04 in a Docker container.

`apt-get install -y libignition-math3 `
`apt-get install -y libsdformat5 \
        libgazebo8 \
        gazebo8 \
        libgazebo8-dev \
        ros-kinetic-ros-control \
        ros-kinetic-ros-controllers`

Then pulled the gazebo_ros package into a catkin_workspace and built.

Any roslaunch of the gazebo_ros package will open Gazebo, and seemingly start up ROS packages, but it won't actually launch the world file. To replicate, install Gazebo as described above and `roslaunch gazebo_ros mud_world.launch`. You'll see nothing is actually in place.

j-rivero, 2017-02-08:
Thanks for the report Ben.

Could you please try to run the standalone gazebo to see if it runs?

If gazebo runs, could you run with `debug:=true` ([instructions here](http://gazebosim.org/tutorials/?tut=ros_roslaunch)) to see if we get any information about what's going on?

BenBlumer, 2017-02-08:
Hi Jose,

Thanks for the quick reply.

`gazebo` runs fine.

`roslaunch gazebo_ros mud_world.launch debug:=true` runs Gazebo, but none of the models of mud_world appear (same for rubble_world, etc.) but the debug tag doesn't provide too much info:
```
... logging to /root/.ros/log/566038dc-ee53-11e6-be81-0242ac110002/roslaunch-9b66e563f9a3-982.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://9b66e563f9a3:42052/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [993]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 566038dc-ee53-11e6-be81-0242ac110002
process[rosout-1]: started with pid [1006]
started core service [/rosout]
process[gazebo-2]: started with pid [1009]
process[gazebo_gui-3]: started with pid [1013]
[ INFO] [1486595239.378860594]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1486595239.379325566]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1486595240.928682559, 0.021000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1486595240.957373940, 0.049000000]: Physics dynamic reconfigure ready.

```

j-rivero, 2017-02-08:
> Thanks for the quick reply.

Thanks for the quick test. Another question: is the standalone gazebo (no ROS) able to load the worlds?

BenBlumer, 2017-02-08:
Yeah: 
`root@9b66e563f9a3:/usr/share/gazebo-8/worlds# gazebo mud.world ` runs Gazebo and sets up the world with the robots and the mud pit fine.

j-rivero, 2017-02-08:
Thanks. Looks like we will need to debug what's going on with the parameter passing mechanism. I need to find some time for this.

BenBlumer, 2017-02-09:
Hey Jose -- do you debug in a Docker environment? If so, here's the docker file to run Ubuntu 16.04/ROS Kinetic/Gazebo 8:

https://github.com/BenBlumer/gazebo8_with_ros_docker

davetcoleman, 2017-02-16:
I know you guys have changed the support cycle of Gazebo + Kinetic: out of curiosity, how many version of Gazebo will the ``kinetic-devel`` branch need to support?

j-rivero, 2017-02-16:
> ow many version of Gazebo will the kinetic-devel branch need to support?

From experience, being an LTS, it is hard to say but we have been supporting in Indigo every single Gazebo version out there.

j-rivero, 2017-03-31:
I tested the current kinetic-devel branch together with gazebo8 and they seems to work pretty well on my system. I've created the binary packages: https://discourse.ros.org/t/gazebo8-ros-kinetic-wrappers-available/1585

@BenBlumer could you confirm that they work for you?

j-rivero, 2017-04-24:
Please reopen if the problem persists

closed 2017-04-24 by j-rivero

BenBlumer, 2017-05-03:
Hi Jose,

I've tested a few install options in Docker and:

- the docker file posted above does not work (roslaunch doesn't load the models specified in the launch file)
- Using the OSRF kinetic file and then installing ros-kinetic-gazebo8-ros-pkgs from the OSRF repository doesn't work. (Error: `The following packages have unmet dependencies:
 ros-kinetic-gazebo8-ros-pkgs : Depends: ros-kinetic-gazebo8-plugins but it is not going to be installed
E: Unable to correct problems, you have held broken packages.`)

I'll try a couple more approaches and update.

BenBlumer, 2017-05-03:
Basically, from an install of ros-kinetic-desktop-full, followed by the [instructions](http://gazebosim.org/tutorials/?tut=ros_wrapper_versions):

- Add the osrfoundation repository to your sources list.
- Install ros-kinetic-gazebo8-ros-pkgs from the osrfoundation repository, which will install the gazebo8 package.

I get a series of errors which seem to boil down to a conflict between ros-kinetic-gazebo8-msgs and ros-kinetic-desktop-full:
`root@826a4c322ef1:/# apt-get install ros-kinetic-gazebo8-ros-pkgs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-gazebo8-ros-pkgs : Depends: ros-kinetic-gazebo8-plugins but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@826a4c322ef1:/# apt-get install ros-kinetic-gazebo8-plugins
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-gazebo8-plugins : Depends: ros-kinetic-gazebo8-msgs but it is not going to be installed
                               Depends: ros-kinetic-gazebo8-ros but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@826a4c322ef1:/# apt-get install ros-kinetic-gazebo8-ros
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-gazebo8-ros : Depends: ros-kinetic-gazebo8-msgs but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@826a4c322ef1:/# apt-get install ros-kinetic-gazebo8-msgs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  debhelper dh-strip-nondeterminism fltk1.3-doc fluid fonts-lato freeglut3-dev gazebo7 gazebo7-common gazebo7-plugin-base gettext gettext-base gfortran gfortran-5 gir1.2-atk-1.0 gir1.2-freedesktop
  gir1.2-gdkpixbuf-2.0 gir1.2-glib-2.0 gir1.2-gtk-2.0 gir1.2-pango-1.0 groff-base intltool-debian libarchive-zip-perl libasprintf-dev libasprintf0v5 libatk1.0-dev libaudio2 libblas-dev
  libbulletcollision2.83.6 libbulletdynamics2.83.6 libcairo-gobject2 libcairo-script-interpreter2 libcairo2-dev libcroco3 libdc1394-22 libfile-stripnondeterminism-perl libflann-dev libflann1.8
  libfltk-cairo1.3 libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev libfreeimage-dev libgazebo7 libgazebo7-dev libgdk-pixbuf2.0-dev libgettextpo-dev libgettextpo0
  libgfortran-5-dev libgirepository-1.0-1 libglib2.0-bin libglib2.0-dev libgtk2.0-dev libgtkglext1 libgts-0.7-5 libgts-bin libgts-dev libharfbuzz-dev libharfbuzz-gobject0 libharfbuzz-icu0
  libignition-math2 libignition-math2-dev liblapack-dev liblinearmath2.83.6 libmail-sendmail-perl libmng2 libopencv-calib3d2.4v5 libopencv-contrib2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5
  libopencv-flann2.4v5 libopencv-highgui2.4v5 libopencv-imgproc2.4v5 libopencv-legacy2.4v5 libopencv-ml2.4v5 libopencv-objdetect2.4v5 libopencv-photo2.4v5 libopencv-video2.4v5 libopenni-dev
  libopenni-sensor-pointclouds0 libopenni0 libpango1.0-dev libpangox-1.0-0 libpangoxft-1.0-0 libpcl-apps1.7 libpcl-common1.7 libpcl-dev libpcl-features1.7 libpcl-filters1.7 libpcl-io1.7 libpcl-kdtree1.7
  libpcl-keypoints1.7 libpcl-octree1.7 libpcl-outofcore1.7 libpcl-people1.7 libpcl-recognition1.7 libpcl-registration1.7 libpcl-sample-consensus1.7 libpcl-search1.7 libpcl-segmentation1.7
  libpcl-surface1.7 libpcl-tracking1.7 libpcl-visualization1.7 libpcl1.7 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpixman-1-dev libproj-dev libprotoc-dev libprotoc9v5 libqt4-dbus libqt4-declarative
  libqt4-designer libqt4-dev libqt4-dev-bin libqt4-help libqt4-network libqt4-opengl libqt4-opengl-dev libqt4-qt3support libqt4-script libqt4-scripttools libqt4-sql libqt4-sql-mysql libqt4-svg
  libqt4-test libqt4-xml libqt4-xmlpatterns libqtcore4 libqtdbus4 libqtgui4 libqtwebkit-dev libqtwebkit4 libraw1394-11 libruby2.3 libsdformat4 libsdformat4-dev libsimbody-dev libsimbody3.5v5 libspnav0
  libsys-hostname-long-perl libtar-dev libtar0 libtbb-dev libtbb2 libtimedate-perl libtinyxml2-2v5 libunistring0 libusb-1.0-0 libusb-1.0-0-dev libusb-1.0-doc libv4l-0 libv4lconvert0 libvtk-java
  libvtk5.10 libxcb-shm0-dev libxcomposite-dev libxcursor-dev libxi-dev libxinerama-dev libxml2-utils libxmu-dev libxmu-headers libxrandr-dev man-db openni-utils po-debconf python-opencv qdbus qt-at-spi
[... cut ...]
  rubygems-integration sdformat-sdf ttf-dejavu-core ttf-liberation unzip x11proto-composite-dev x11proto-randr-dev x11proto-xinerama-dev zip
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  ros-kinetic-desktop-full ros-kinetic-gazebo-msgs ros-kinetic-gazebo-plugins ros-kinetic-gazebo-ros ros-kinetic-gazebo-ros-pkgs ros-kinetic-simulators
The following NEW packages will be installed:
  ros-kinetic-gazebo8-msgs
0 upgraded, 1 newly installed, 6 to remove and 272 not upgraded.
Need to get 119 kB of archives.
After this operation, 10.7 MB disk space will be freed.
Do you want to continue? [Y/n] 
`

BenBlumer, 2017-05-04:
OK, so you can get it to install by installing gazebo8-msgs first. Sample Dockerfile attached. 
[Dockerfile.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/975001/Dockerfile.txt)

But unfortunately, `roslaunch gazebo_ros mud_world.launch` still brings up an empty world (and same for launching `rubble_world`, `elevator_world`, etc.)

j-rivero, 2017-05-04:
> I get a series of errors which seem to boil down to a conflict between ros-kinetic-gazebo8-msgs and ros-kinetic-desktop-full:

Hello Ben. The problem seems related to a conflict between the `ros-kinetic-gazebo-*` packages (using azebo7`) and the installation of the `ros-kinetic-gazebo8-*` packages that apt is not able to solve.

Solutions could be:
 * Remove any gazebo package in the system before installing `ros-kinetic-gazebo8-`
 * Use `aptitude` instead of `apt` and check if it is able to solve the conflict.

> But unfortunately, roslaunch gazebo_ros mud_world.launch still brings up an empty world (and same for launching rubble_world, elevator_world, etc.)

They are working on my Xenial system (no docker, just plain system). You could be facing a problem related to docker interaction with the X system. I highly recommend to check out [the ROS page related to this topic](http://wiki.ros.org/docker/Tutorials/GUI)

BenBlumer, 2017-05-09:
Hi Jose,

I don't believe it's related to the X system because my system works fine with Gazebo 7 in docker (see attached) thanks to the methods posted on the page you linked to.
 
To solve the conflicts, I've tried starting from kinetic-ros-base and installing Gazebo 8 from the OSRF repo. Indeed, there are no conflicting packages anymore, but Gazebo crashes when, and only when, I roslaunch it.

I've attached the message from the crash. Does this mean anything to you?

[Dockerfile_gazebo8.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/987388/Dockerfile_gazebo8.txt)

[Dockerfile_gazebo7.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/985112/Dockerfile_gazebo7.txt)

[roslaunch_crash.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/987397/roslaunch_crash.txt)

BenBlumer, 2017-05-09:
Anddddd ... To make things more interesting: 

If one first launches gzclient 
`gzclient &`
And then roslaunches, e.g., mudworld:
`roslaunch gazebo_ros mud_world.launch`

you still get the same crash log (big stack trace followed by `gazebo_gui-3 ... exit code 134 ...` but mud_world launches and seems to work normally.  

I figure this has to be something about the way that ros_gazebo is interacting with ros or gazebo. Perhaps something related to this: http://answers.gazebosim.org/question/4252/gazebo_gui-3-error-when-starting-gazebo-with-roslaunch/?answer=12622#post-id-12622

jacknlliu, 2017-05-29:
@j-rivero Did this issue have been resolved? And could we use `ros-kinetic-gazebo8-*` with other ROS packages excluding `gazebo`?

andre-nguyen, 2017-05-29:
hhhm I'm getting error trying to run gazebo 8 with ROS of the sort:
```
roslaunch gazebo_ros mud_world.launch debug:=true
... logging to /home/inflatablequad/.ros/log/d31c6f5e-447e-11e7-80d2-2c4d54d78295/roslaunch-L5816-10-20197.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://L5816-10:46125/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [20208]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d31c6f5e-447e-11e7-80d2-2c4d54d78295
process[rosout-1]: started with pid [20221]
started core service [/rosout]
process[gazebo-2]: started with pid [20227]
process[gazebo_gui-3]: started with pid [20241]
[ INFO] [1496069716.070887931]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1496069716.092111442]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1496069717.591603108, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1496069717.610724956, 0.040000000]: Physics dynamic reconfigure ready.
*** Error in `gzclient': double free or corruption (out): 0x0000000005f33500 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f1706dc37e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7f1706dcbe0a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f1706dcf98c]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_+0xaf)[0x7f170744a3ef]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(+0x1ae0ff)[0x7f17089ee0ff]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(_ZN6gazebo3gui12RenderWidget9AddPluginERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10shared_ptrIN3sdf7ElementEE+0xc6)[0x7f1708a40b86]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(_ZN6gazebo3gui12RenderWidget10AddPluginsERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS8_EE+0xf6)[0x7f1708a410a6]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(_ZN6gazebo3gui3runEiPPc+0x146)[0x7f17089b2a16]
gzclient[0x40293c]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f1706d6c830]
gzclient(_start+0x29)[0x402ba9]
```

Anyone else?

j-rivero, 2017-05-29:
> @j-rivero Did this issue have been resolved? And could we use ros-kinetic-gazebo8-* with other ROS packages excluding gazebo?

The issue is resolved in the sense that we have now kinetic packages supporting `gazebo8`

> hhhm I'm getting error trying to run gazebo 8 with ROS of the sort:

This has nothing to do with the original goal of the issue and the problem was introduced in `gazebo 8.1.0 `.  See #581