ros-simulation/gazebo_ros_pkgs #548: The time stamp in the header of the message camera/depth/image_raw is occasionally set to zero.

renweizhukov, 2017-02-22:
Run one demo given in the turtle_gazebo tutorial at http://wiki.ros.org/turtlebot_gazebo/Tutorials/indigo/Explore%20the%20Gazebo%20world.

```bash
$ roslaunch turtlebot_rviz_launchers view_robot.launch
```

The time stamp is the header of the message camera/depth/image_raw is expected to monotonically increasing but occasionally is set to zero as shown in the attached image.

[image]

I have shared the ROS bag at https://1drv.ms/u/s!Aq1vY_5CN_PkqIwA8C80hF8Mdj8icA

iche033, 2017-02-22:
a few pull requests (#410, #540) that fix this problem were merged recently but they're target at kinetic and jade. I don't know if it will be ported to indigo though. @j-rivero

renweizhukov, 2017-02-23:
Thank iche033 for the info. I was actually running the demo on ROS kinetic but I installed the Ubuntu official gazebo_ros package via "apt install", so I guess that it may not contain the fix, right? Do you know when the fix will make the way to the official package?

iche033, 2017-02-23:
I see, yes it's likely that a new version has not been released yet. I'm not sure it'll be done. It'll be up to the maintainers.

j-rivero, 2017-02-23:
The new version 2.5.9 of gazebo_ros_pkgs for kinetic [was released](https://github.com/ros/rosdistro/pull/13934) some days ago. Those packages are ready in [the shadow repository](http://wiki.ros.org/ShadowRepository) and will be updated with the next sync [that will be announced here](https://discourse.ros.org/c/release/kinetic)

renweizhukov, 2017-02-23:
Thank j-rivero for the info. Look forward to the kinetic next sync.

ironmig, 2018-06-18:
Duplicate of #490 (fixed).  Comment to re open this issue if it still occurs on supported ROS distros.

closed 2018-06-18 by ironmig