ros-simulation/gazebo_ros_pkgs #559: Silent failure on launch if plugin not found

rconnorlawson, 2017-04-09:
When launching URDF models via roslaunch with the spawn_models script, it is possible for plugins to silently fail to initialize. 

If the plugin library is not present, the plugin will not initialize, and Gazebo will issue no warnings or logs that this has occurred. The model will appear in the Gazebo world, and the script will complete cleanly, but the plugin behavior will not activate.

I observed this behavior using the following command in roslaunch:

    <node name="spawn_turtlebot_model" pkg="gazebo_ros" type="spawn_model"
        args="$(optenv ROBOT_INITIAL_POSE) -unpause -urdf -param robot_description -x 2 -y 3 -z 0 -Y 3.14 -model mobile_base"/>

In particular, somewhere in the URDF `libgazebo_ros_kobuki.so` was referenced as a plugin. In my workflow I was building this library from source. I had recently cleaned my build, so the library was missing.

With the library is unavailable, the output from roslaunch was:

    [ INFO] [1491772598.999104167]: Finished loading Gazebo ROS API Plugin.
    [ INFO] [1491772598.999515277]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
    [ INFO] [1491772599.705111397, 999.937000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
    [ INFO] [1491772600.159808146, 1000.074000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
    [ INFO] [1491772600.163509518, 1000.074000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
    [ INFO] [1491772600.213022476, 1000.074000000]: Physics dynamic reconfigure ready.

Note that the Kobuki plugin is never mentioned in the output. The issue here is that **there is no warning that the Kobuki plugin could not be loaded**, or that it attempted to load the Kobuki plugin at all. An acceptable solution is simply to issue a ROS warning when a plugin fails to load.

The "workaround" is to know up front what plugins you expect to load, and make sure the output shows their initialization. However, this is unsatisfying at best and very difficult at worse. Presupplied URDF packages (such as turtlebot_description) are often difficult to parse by hand, and don't come with documentation about what plugins they expect.

Environment specs:
ROS Indigo
Ubuntu 14.04
Gazebo 7.6

j-rivero, 2017-04-10:
Failing to load a plugin should leave an error in one of the logs inside .gazebo/ , probably with something like `Failed to load plugin`. If you want to enable the error output in the command line, you could send the `--verbose` flag to the gazebo invocation from roslaunch, something like [what this example](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros/launch/empty_world.launch#L15) does.

rconnorlawson, 2017-04-10:
Thank you for the quick response, this is exactly what I needed. Using the verbose argument did generate the output I was looking for. 

Is there documentation for the verbose setting/other output options? I'm not sure how verbose ended up turned off in my stack, but regardless it seems like disabling warnings and errors while keeping only info is a weird setting. I would expect the opposite - show warnings and errors always, and only show info/msg in verbose mode.

j-rivero, 2017-04-10:
The issue of enabling the verbose flag by default has been discussed by the Gazebo team different times. I think that the current consensus is to [implement levels of warnings](https://bitbucket.org/osrf/gazebo/issues/1705/add-optional-level-to-verbose-flag). So nothing just turned off your flag, it is off by default. We could consider to enable it by default for `gazebo_ros_pkgs` although I prefer not to change the default experience from using Gazebo upstream to use the ROS wrappers on it.

About the documentation of this verbose feature I think that it is missing from the [ROS tutorials](http://gazebosim.org/tutorials?cat=connect_ros). There is some information about [using roslaunch](http://gazebosim.org/tutorials?tut=ros_roslaunch&cat=connect_ros) but nothing I can found for the verbose flag.

ironmig, 2018-06-15:
Closing as the originaly described issue is not a bug. I added documentation about the verbose option to the tutorials in [a pr](https://bitbucket.org/osrf/gazebo_tutorials/pull-requests/512/add-verbose-flag-description-to/diff)

closed 2018-06-15 by ironmig