ros-simulation/gazebo_ros_pkgs #560: Laser Plugin throws an Exception

canalteks, 2017-04-10:
Dear developers,

I'm using a virtual laser sensor with gazebo_ros_laser plugin (**non GPU** version) in ROS kinetic and gazebo7, Debian jessie.
Unfortunatelly,  gazebo_ros_**gpu**_laser plugin (which I do **not** load) throws an exception `GazeboRosLaser controller requires a Ray Sensor as its parent`, saying after `GpuRayPlugin not attached to a GpuLaser sensor`.

For a workaround, I found the following solution to the problem
`
cd /opt/ros/kinetic/lib/
`
`
mv libgazebo_ros_gpu_laser.so  notused_libgazebo_ros_gpu_laser.so 
`

I thought that the following items would cause that error
- [gazebo_ros_laser](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_laser.cpp#L46) and [gazebo_ros_gpu_laser](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_gpu_laser.cpp#L47) have been registerd as a same object by `GZ_REGISTER_SENSOR_PLUGIN(GazeboRosLaser)` in their source codes
- we expect that gazebo loads (gazebo_ros_laser or gazebo_ros_gpu_laser) plugin which we set in URDF, but not.

I can reproduce the error along the following sequence
1. loading a gazebo world file **embedded with some model plugins (their contents could be empty)**
2. spawn a robot model with **gazebo_ros_laser** plugin via gazebo_ros::spawn_model
3. gazebo throws an exception

I do not know but the above sequence runs safely in ROS indigo and gazebo.

package version information (the system was updated yesterday)
- gazebo7 7.0.0+dfsg-2osrf1~jessie1
- ros-kinetic-gazebo-ros 2.5.10-0jessie-20170308-224905-0800
- ros-kinetic-gazebo-plugins 2.5.10-0jessie-20170327-151102-077

To avoid this error, I suggest that
- change the name `GazeboRosLaser` to `GazeboRosGpuLaser` for GPU
- `#define GazeboRosLaser GazeboRosGpuLaser` in its header file
- use `GZ_REGISTER_SENSOR_PLUGIN(GazeboRosGpuLaser)` in its source file

I hope it could be helpful.

j-rivero, 2017-04-10:
Hello Saida:

Thanks very much for the complete bug report. Your diagnostic sounds reasonable to me. Would you mind creating a PR with your proposed changes against the kinetic-devel branch?

canalteks, 2017-04-11:
@j-rivero
Hello, Rivero.
Thank you for your quick reply.
There is no problem to use my report.
I'm always grateful to gazebo_ros team.

Thanks.

closed 2017-04-11 by canalteks

shubham-shahh, 2022-05-26:
Hi @canalteks, @j-rivero is the issue resolved?