ros-simulation/gazebo_ros_pkgs #562: New Kinetic Release?

davetcoleman, 2017-04-12:
Now that https://github.com/ros-simulation/gazebo_ros_pkgs/pull/556 is merged back in, we should release this package to allow maintainers time to fix their CMakeLists

j-rivero, 2017-04-18:
You are right, here we go https://github.com/ros/rosdistro/pull/14563

closed 2017-04-18 by j-rivero

davetcoleman, 2017-04-21:
Where can I find a list of Gazebo-dependent packages that are now broken on the build farm because of this change? http://repositories.ros.org/status_page/ros_kinetic_default.html?q=gazebo ?

I can contact them if any exist...

j-rivero, 2017-04-21:
I don't have any previous experience looking for these kind of breakages but looking into the regressions of kinetic I think that can get a hint: http://repositories.ros.org/status_page/ros_kinetic_default.html?q=REGRESSION

j-rivero, 2017-04-24:
@davetcoleman we have a problem related to the change https://github.com/yujinrobot/kobuki_desktop/issues/50 . I don't have too much time now to look deeply how to solve it, if you want can provide a patch that would be great. Otherwise, we would need to revert the change (once again).

davetcoleman, 2017-04-25:
I don't have time to fix other robot's that use Gazebo... I got it working for Baxter with my fix, too bad it has to be reverted yet again.

j-rivero, 2017-04-25:
> I don't have time to fix other robot's that use Gazebo... I got it working for Baxter with my fix, too bad it has to be reverted yet again.

sorry for that, I was unable to find any other solution to unblock the kinetic release yesterday. Let's try to finally merge the [gazebo-dev](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/571) package to solve the warning and implement a better solution for third party software.