ros-simulation/gazebo_ros_pkgs #565: ROS Lunar Release

j-rivero, 2017-04-19:
In order to prepare the upcoming Lunar Release I'm going to create the lunar-devel branch. `gazebo8` is the version selected to be present in Lunar.

If any of you have a special interest for unstable code that was not merged and is waiting in a PR, this is a good moment to request the inclusion into the Lunar branch. Disruptive changes, code clean up and other changes (like those affecting the names of ROS Topics) are also welcome.

ROS will drop the support for Jade when the new new Lunar release is out, so we probably want to follow it an drop it from this repository.

j-rivero, 2017-04-19:
From the top of my head, pull #473 will be definitely in since the ROS Lunar initial release (although it does not mean that we consider it for other active branches).

mikaelarguedas, 2017-04-25:
@j-rivero FYI: all dependencies of gazebo_ros_pkgs are [now satisfied](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22gazebo_ros_pkgs%22&s=6&r=1), the gazebo debs have been [pulled in the repos](https://github.com/ros-infrastructure/reprepro-updater/pull/40#issuecomment-297084612) and the rosdep keys [have been merged](https://github.com/ros/rosdistro/pull/14564)

j-rivero, 2017-04-25:
> @j-rivero FYI: all dependencies of gazebo_ros_pkgs are now satisfied, the gazebo debs have been pulled in the repos and the rosdep keys have been merged

Thanks Mikael. I'm in the process of merging the last pull requests against kinetic-devel, fork the branch for lunar and prepare the first release.

j-rivero, 2017-04-27:
The fixes for compilation and warnings are ready in pull request #573

j-rivero, 2017-04-28:
first try https://github.com/ros/rosdistro/pull/14712

mikaelarguedas, 2017-04-29:
@j-rivero FYI: these packages [are failing on ARM](https://github.com/ros/rosdistro/pull/14712#issuecomment-297891874)

mikaelarguedas, 2017-05-17:
@j-rivero Lunar [now targets gazebo 7](https://discourse.ros.org/t/lunar-to-target-gazebo-7/1840/1), can you make a new release of this repo with rosdep keys updated to link against Gazebo 7 ?
Thanks!

j-rivero, 2017-05-17:
rosdistro pull request in place https://github.com/ros/rosdistro/pull/15000

mikaelarguedas, 2017-05-19:
@j-rivero is is possible to make a new release of this repository using the updated rosdep keys ?

j-rivero, 2017-05-19:
> @j-rivero is it possible to make a new release of this repository using the updated rosdep keys ?

I'm checking what happen in the lunar branch after migrating it to gazebo8. We probably need back to use the code in kinetic-devel. We should be able to have a new release soon, before Monday.

j-rivero, 2017-05-19:
pull request #583 in place. Waiting for the CI scripts to finish before merging and releasing.

j-rivero, 2017-05-19:
> Waiting for the CI scripts to finish before merging and releasing.

One little change more https://github.com/ros/rosdistro/pull/15013 . If merged I can try to run bloom during the weekend so we get the packages ready for Monday.

mikaelarguedas, 2017-05-20:
https://github.com/ros/rosdistro/pull/15013 has been merged

j-rivero, 2017-05-21:
New release using gazebo7 is ready: https://github.com/ros/rosdistro/pull/15017

mikaelarguedas, 2017-10-14:
Thanks @j-rivero for getting this released in Lunar on time!
I believe this issue can now be closed

closed 2018-05-25 by j-rivero