ros-simulation/gazebo_ros_pkgs #575: ‘LightPtr’ is not a member of ‘gazebo::physics’

wangzhaodong123, 2017-04-29:
When I catkin_make my work_space , the problem happened:
‘LightPtr’ is not a member of ‘gazebo::physics’
why? can you help me ? Thank you .

j-rivero, 2017-05-04:
I think that your facing a problem migration from one gazebo version to another, probably from gazebo8 to gazebo9. There is a good document in [gazebo upstream project](https://bitbucket.org/osrf/gazebo/src/default/Migration.md?fileviewer=file-view-default) describing the deprecations and the replacement functions.

For this kind of questions related to fails in compilation, please use http://answers.gazebosim.org.

closed 2017-05-04 by j-rivero

wangzhaodong123, 2017-05-06:
Thanks for advise.After installing gazebo7, I met another problem when i catkin_make my work_space:
Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake
Could you tell me why? Thank you.

scpeters, 2017-05-06:
@wangzhaodong123 did you install `libgazebo7-dev`? I believe that has the cmake config files.

wangzhaodong123, 2017-05-06:
@scpeters @j-rivero Thank you very much,and i solved the problem just now.Thank you.