ros-simulation/gazebo_ros_pkgs #578: Crash due locateROI function

pulver22, 2017-05-05:
I am encountering the following error either with Gazebo2 on Indigo and Gazebo 7 on Kinetic. I also downloaded Gazebo 7.7 from the osrf repository without success. It happens randomly when running simulation with an high real time factor (~>3). 

`OpenCV Error: Assertion failed (dims <= 2 && step[0] > 0) in locateROI, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/matrix.cpp, line 949
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/matrix.cpp:949: error: (-215) dims <= 2 && step[0] > 0 in function locateROI

Aborted (core dumped)
`

I really don't know how to solve the issue. Just to mention that I don't call that locateROI function at all.

j-rivero, 2017-05-05:
Sounds like a problem related to the a race condition somehow. To debug it, we would need:

* The backtrace from the core dump
* A repository of code with a test case that it is able to reproduce the problem.

pulver22, 2017-05-07:
From the log generated in .ros/log, the error reported is the following:

```
[rospy.internal][WARNING] 2017-05-04 01:31:30,036: Unknown error initiating TCP/IP socket to Workstation:60881 (http://Workstation:41479/): Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/network.py", line 362, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

[rospy.internal][INFO] 2017-05-04 01:31:30,037: topic[/clock] removing connection to http://Workstation:41479/
[rospy.internal][WARNING] 2017-05-04 01:31:30,037: Unknown error initiating TCP/IP socket to Workstation:60881 (http://Workstation:41479/): Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/network.py", line 362, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

```

For the repository, I will create a proper test-set for reproducing the error. But I have to say that it's impossible to predict in advance when the crash occurs.

pulver22, 2017-05-10:
Hi @j-rivero, I think I was able to find the proper dump file (Sorry, It's the first time I do it!)
You can download it from [here](https://liveplymouthac-my.sharepoint.com/personal/riccardo_polvara_plymouth_ac_uk/_layouts/15/guestaccess.aspx?docid=11697f4fc1f92419d82cf3646b1323d98&authkey=AW4mSnAsuyKpzmE5U9pgq3o)

j-rivero, 2017-05-10:
Thanks. Looking at the logs looks to me that there is a problem in the network layer of ROS, similar to [this one](http://answers.ros.org/question/11488/unknown-error-initiating-tcpip-socket/). Hard to diagnose.

The attach file is a .crash? Are you working on Linux? Seems like it is not a core dump.
`
"/home/jrivero/Downloads/_usr_bin_gzserver-7.7.0.1000.crash" is not a core dump: File format not recognized
`

pulver22, 2017-05-10:
Yes, Ubuntu 16.04 . I tried to generate a code dump setting `ulimit -c unlimited` but no files have been generated when gzserver crashed. The one I attached is the only one I found in /var/crash. Sorry, I've never done it before and probably I did something wrong.