ros-simulation/gazebo_ros_pkgs #581: Gazebo 8.1 segfault

GertKanter, 2017-05-16:
Is ABI broken in 8.1?

Upgrading from Gazebo 8.0 to 8.1 and we're getting segfaults every time we run

`rosrun gazebo_ros gzclient --verbose`

`[Err] [RenderWidget.cc:319] System is attempting to load a plugin, but detected an incorrect plugin type. Plugin filename[/opt/ros/kinetic/lib/libgazebo_ros_paths_plugin.so].
Segmentation fault (core dumped)`

or something uglier like

`*** Error in 'gzclient': double free or corruption (out): 0x0000000005317400 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f3f142a57e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7f3f142ade0a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f3f142b198c]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_+0xaf)[0x7f3f1492c3ef]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(+0x1ae0ff)[0x7f3f15ed00ff]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(_ZN6gazebo3gui12RenderWidget9AddPluginERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10shared_ptrIN3sdf7ElementEE+0xc6)[0x7f3f15f22b86]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(_ZN6gazebo3gui12RenderWidget10AddPluginsERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS8_EE+0xf6)[0x7f3f15f230a6]
/usr/lib/x86_64-linux-gnu/libgazebo_gui.so.8(_ZN6gazebo3gui3runEiPPc+0x146)[0x7f3f15e94a16]
gzclient[0x40293c]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f3f1424e830]
gzclient(_start+0x29)[0x402ba9]
======= Memory map: ========
00400000-00a22000 r-xp 00000000 fd:01 3151860                            /usr/bin/gzclient-8.1.0
00c21000-00c22000 r--p 00621000 fd:01 3151860                            /usr/bin/gzclient-8.1.0
00c22000-00c23000 rw-p 00622000 fd:01 3151860                            /usr/bin/gzclient-8.1.0
02193000-054fa000 rw-p 00000000 00:00 0                                  [heap]
`

j-rivero, 2017-05-16:
> Is ABI broken in 8.1?

I was able to reproduce the error. I'm working on it.

j-rivero, 2017-05-16:
I submitted an issue to the [gazebo issue tracker](https://bitbucket.org/osrf/gazebo/issues/2279/core-dump-with-gazebo-81-in).

onlytailei, 2017-05-23:
Same question

j-rivero, 2017-05-23:
The issue is still open in [gazebo issue tracker](https://bitbucket.org/osrf/gazebo/issues/2279/core-dump-with-gazebo-81-in) and it is consider a regression in the gazebo8 series, so it needs to be fixed there. Contributions to gazebo code that fix the plugin problem are very welcome and could speed up the process, the details are in the issue.

Meantime, I'm afraid that the workaround is to uninstall `gazebo8 8.1.0` and install  `gazebo8 8.0.0` from packages. All the 8.0.0 packages are available in from [gazebo distributions page](http://gazebosim.org/distributions/gazebo/releases/) (give the page 5 minutes to load, it takes a while).

onlytailei, 2017-05-23:
@j-rivero  yes, recompile works

andre-nguyen, 2017-05-29:
The gazebo distributions page takes so long to long I wasn't sure it would actually do it.

Direct link to the tar ball of 8.0.0 for anyone interested http://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-8.0.0.tar.bz2

j-rivero, 2017-05-29:
> The gazebo distributions page takes so long to load I wasn't sure it would actually do it

Give it time, go for a coffee. Note that in that you also have all the gazebo8 debian packages in the case that you don't want to compile the code from source.

The original problem should be fixed with this [PR in gazebo bitbucket](https://bitbucket.org/osrf/gazebo/pull-requests/2716/added-gui-client-plugin-option-leave-g-to/diff). Feel free to test it and report the result on the PR page so we can speed up things and release a new 8.1.1. Thanks!

andre-nguyen, 2017-05-30:
works for me! Commented on bitbucket too

j-rivero, 2017-06-05:
8.1.1 is out in the stable repository. It should fix the problem. Thanks everyone for the report and help. Please reopen if it is not fixed for you.

closed 2017-06-05 by j-rivero