ros-simulation/gazebo_ros_pkgs #587: catkin error gazebo-ros-control missing on indigo plus gazebo7

Shawnnn, 2017-05-30:
I install gazebo7 and then install gazebo7-ros-pkgs from debian as [this  instruction](http://answers.gazebosim.org/question/16092/remove-gazebo2-and-install-gazebo7-for-ros-indigo/)

I download code from github (tutorial code from book, mastering ros for robotics programming), and then catkin, it throws build error as follows:

    CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
      Could not find a package configuration file provided by
      "gazebo_ros_control" with any of the following names:
    
        gazebo_ros_controlConfig.cmake
        gazebo_ros_control-config.cmake

 

     Add the installation prefix of "gazebo_ros_control" to CMAKE_PREFIX_PATH or
      set "gazebo_ros_control_DIR" to a directory containing one of the above
      files.  If "gazebo_ros_control" provides a separate development package or
      SDK, be sure it has been installed.
    Call Stack (most recent call first):
      seven_dof_arm_gazebo/CMakeLists.txt:7 (find_package)
    
    
        -- Could not find the required component 'gazebo_ros_control'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
        CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
    
      Could not find a package configuration file provided by
    
          "gazebo_ros_control" with any of the following names:
        
            gazebo_ros_controlConfig.cmake
            gazebo_ros_control-config.cmake
    
      Add the installation prefix of "gazebo_ros_control" to CMAKE_PREFIX_PATH or
      set "gazebo_ros_control_DIR" to a directory containing one of the above
      files.  If "gazebo_ros_control" provides a separate development package or
      SDK, be sure it has been installed.
    Call Stack (most recent call first):
      seven_dof_arm_gazebo/CMakeLists.txt:7 (find_package)

It seems like I have not installed gazebo-ros-control package, though I think it is supposed to have been installed already by   
previous install procedure:

    sudo apt-get install ros-indigo-gazebo7-ros-pkgs

I check the packages by:

    rospack list | grep gazebo_

whose result is :

    gazebo_msgs /opt/ros/indigo/share/gazebo_msgs
    gazebo_plugins /opt/ros/indigo/share/gazebo_plugins
    gazebo_ros /opt/ros/indigo/share/gazebo_ros
    gazebo_tutorials /home/shawn/catkin_ws/src/gazebo_ros_demos/custom_plugin_tutorial

**It turns out gazebo-ros-control package haven't been installed! Why?**

According to instruction of gazebo official, it is said that except for ros-indigo-gazebo7-ros-pkgs, all other packages related to gazebo MUST be built from source.

Does it mean I have to build gazebo-ros-control from source?

Is it version-specific? Where can I find the right version on github?

[Is this the right link?](https://github.com/ros-simulation/gazebo_ros_pkgs/tree/kinetic-devel/gazebo_ros_control), if so, how can I git clone this package on this page?

j-rivero, 2017-05-30:
> It turns out gazebo-ros-control package haven't been installed! Why?

Because the `gazebo-ros-pkgs` metapackage does not depend on them. I think that [the official documentation](http://gazebosim.org/tutorials?tut=ros_installing) is clear on the command line to install them. The reasons why gazebo-ros-control is not part of the metapackage is probably because `gazebo_ros_control` pulls many dependencies and it was not consider as part of the usual use case for the ROS users of Gazebo. Thus, it is available as a package but not part of the metapackage. 

> According to instruction of gazebo official, it is said that except for ros-indigo-gazebo7-ros-pkgs, all other packages related to gazebo MUST be built from source.
> Does it mean I have to build gazebo-ros-control from source?

Nop. Try just using `sudo apt-get install ros-indigo-gazebo7-ros-control` and you are ready. What that means is that you can not use any ROS released package for Indigo which use or depend on gazebo or gazebo_ros_pkgs (i.e. `hector_gazebo_plugins`, ...)  . You will need to build them from source.

closed 2017-05-30 by j-rivero

Shawnnn, 2017-05-31:
Thanks for your prompt reply and helpful instruction. It works now.
I saw the official link you shared, but it is said that:

> These instructions are for using the Gazebo versions that are fully integrated with ROS Kinetic, ROS Jade and ROS Indigo.

Thereby I worried that I couldn't do this way cause the gazebo7 is not corresponding to indigo, and that's why I didn't try:
`sudo apt-get install ros-indigo-gazebo7-ros-control`

j-rivero, 2017-05-31:
> Thereby I worried that I couldn't do this way cause the gazebo7 is not corresponding to indigo, and that's why I didn't try:

Makes sense, thanks for the feedback. I've sent a change for the tutorial to make those packages explicit: https://bitbucket.org/osrf/gazebo_tutorials/pull-requests/431/explicit-the-installation-of-ros-control/diff

Hopefully it can help to avoid the problem in the future.

supasani6, 2017-07-11:
Hi, I tried to install gazebo_ros_control using 'sudo apt-get install ros-kinetic-gazebo7-ros-control' , but it says 
E: Unable to locate package ros-kinetic-gazebo7-ros-control
I am using ROS kinetic on Ubuntu 16.04 which by default uses Gazebo7.0.0
Could you tell me what the problem could be?

j-rivero, 2017-07-11:
Kinetic is using `gazebo7` so you can get the gazebo7 version of ros control just by: sudo apt-get install  ros-kinetic-gazebo-ros-control .

varunvp, 2018-01-19:
> Hi, I tried to install gazebo_ros_control using 'sudo apt-get install ros-kinetic-gazebo7-ros-control' , but it says
> E: Unable to locate package ros-kinetic-gazebo7-ros-control

Well, same problem, except that I'm running ROS Indigo and Gazebo 7. I changed the commands accordingly. When I found this git, it was a huge relief 
https://github.com/osrf/gazebo7_ros_pkgs-release/tree/debian/indigo/gazebo_ros_pkgs
But I ran into difficulties cloning the other branches in debian/indigo/ which are dependencies for gazebo_ros_pkgs. Apparently you can't clone multiple branches into one directory. Also, my knowledge of git and cloning is limited.

I miss you, Windows :-(

varunvp, 2018-01-19:
So I cloned each branch in https://github.com/osrf/gazebo7_ros_pkgs-release/tree/debian/indigo/ in seperate folders. After running rosdep, I was shocked to find that it was listing gazebo 2 as a missing dependency! That too, in a gazebo7 package!!

varunvp, 2018-01-20:
> I don't know the details about your environment, but ros-indigo-gazebo7-ros-controlis being successfully installed by our daily testing job: https://build.osrfoundation.org/job/ros_gazebo7_pkgs-install_pkg_indigo-trusty-amd64/274/consoleFull

I am running Ubuntu 14.04 32 bit on an Intel Celeron 900 processor. Does this make a difference?

> I would recommend do not play with that repository unless you know well git and bloom. It is probably going to give you more headaches than solutions.

I happily won't.

> I read the messages coming from issues in this repository but I don't always have time to answer quick. You already sent a comment 3 hours before this last one, please be patient.

I just wanted to let everyone know of the different alternatives that I've tried before asking a question here. I didn't want an immediate response.

j-rivero, 2018-01-22:
> I am running Ubuntu 14.04 32 bit on an Intel Celeron 900 processor. Does this make a difference?

Umm. which is the output of `uname -a`

varunvp, 2018-01-23:
This is the output
`Linux varun-Aspire-5334 4.4.0-109-generic #132~14.04.1-Ubuntu SMP Tue Jan 9 21:45:52 UTC 2018 i686 i686 i686 GNU/Linux`

j-rivero, 2018-01-23:
The processor is 64 bits but the system is 32 bits. I'm afraid that we don't have 32 bits packages for `gazebo7_ros_pkgs`. I would suggest to use a 64 bits system unless you have a good reason to stick with 32bits.

varunvp, 2018-01-24:
Any way I can build one from source for 32 bit?

wajeehulhassanvii, 2019-03-20:
Hi! I am having this error. Can you please help me get through this. I am new to ROS.

```
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /home/wajeeh-ros/hector_quadrotor_tutorial/devel/share/hector_quadrotor_model/cmake/hector_quadrotor_modelConfig.cmake:148 (message):
  Project 'hector_quadrotor_gazebo_plugins' tried to find library
  '-lpthread'.  The library is neither a target nor built/installed properly.
  Did you compile project 'hector_quadrotor_model'? Did you find_package() it
  before the subdirectory containing its code is included?
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/wajeeh-ros/hector_quadrotor_tutorial/build/CMakeFiles/CMakeOutput.log".
See also "/home/wajeeh-ros/hector_quadrotor_tutorial/build/CMakeFiles/CMakeError.log".

```