ros-simulation/gazebo_ros_pkgs #589: gazebo_plugins: diff_drive plugin transform coupled to odom publishing

tcassell94, 2017-06-07:
The source code for the gazebo_ros_diff_drive.cpp shows that it is impossible to publish an odometry message without also publishing a transform. There are two related parameters that are defined in the XML plugin specification: publishTf and publishWheelTF. The publishTf parameter is meant to allow odom topic publishing whilst the publishWheelTF parameter is meant to allow transform sending. However, the function that sends the odom topic message also sends a transform, irrespective of the value of publishWheelTF. If both parameters are true, twice as many transforms are sent. The fix is simple: remove lines 458 - 461 and the parameters will work as designed.

j-rivero, 2017-06-07:
Thanks for the comment. Would you mind please sending a pull request with the code change?

bhazza, 2017-07-20:
I think I disagree with OP - the publishTf parameter is not meant to publish odom topic, it is meant to publish the world/odom pose transform, i.e. the current behaviour. The publishWheelTF parameters publishes a wheel transform, which is not the same.

On a related issue - and the one that lead me to this post - setting publishTf to false also stops the odometry from being published, which I don't think is the intended behaviour.

chapulina, 2018-08-29:
So as I understand it, the [current behaviour](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_diff_drive.cpp) is:

* `<publishTf>`
    * Defaults to true
    * If true, advertize `nav_msgs::Odometry` on `/odom` and publish it
    * If both this and `publishOdomTf` are true, send base-to-odom transforms

* `<publishWheelTF>`
    * Defaults to false
    * If true, send wheel-to-base transforms

* `<publishOdomTF>`
    * Defaults to true
    * If both this and `publishTf` are true, send base-to-odom transforms

My plan while migrating to ROS 2 is to clean this up as follows:

* `<publish_odom>`
    * Defaults to false
    * If true, advertize `nav_msgs::Odometry` on `/odom` and publish it

* `<publish_wheel_tf>`
    * Defaults to false
    * If true, send wheel-to-base transforms

* `<publish_odom_tf>`
    * Defaults to false
    * If true, send base-to-odom transforms

Please speak up if you see a problem with this.