ros-simulation/gazebo_ros_pkgs #594: gazebo_plugins ImportError: No module named msg

Alessandro-Barbieri, 2017-06-24:
building from source on gentoo

```
Scanning dependencies of target gazebo_plugins_gencfg
[ 30%] Generating dynamic reconfigure files from cfg/CameraSynchronizer.cfg: /home/ale/catkin_ws/devel_isolated/gazebo_plugins/include/gazebo_plugins/CameraSynchronizerConfig.h /home/ale/catkin_ws/devel_isolated/gazebo_plugins/lib/python2.7/site-packages/gazebo_plugins/cfg/CameraSynchronizerConfig.py
Generating reconfiguration files for CameraSynchronizer in gazebo_plugins
Wrote header file in /home/ale/catkin_ws/devel_isolated/gazebo_plugins/include/gazebo_plugins/CameraSynchronizerConfig.h
[ 32%] Generating dynamic reconfigure files from cfg/GazeboRosCamera.cfg: /home/ale/catkin_ws/devel_isolated/gazebo_plugins/include/gazebo_plugins/GazeboRosCameraConfig.h /home/ale/catkin_ws/devel_isolated/gazebo_plugins/lib/python2.7/site-packages/gazebo_plugins/cfg/GazeboRosCameraConfig.py
Traceback (most recent call last):
  File "/home/ale/catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/cfg/GazeboRosCamera.cfg", line 6, in <module>
    from dynamic_reconfigure.msg import SensorLevels
ImportError: No module named msg
make[2]: *** [CMakeFiles/gazebo_plugins_gencfg.dir/build.make:88: /home/ale/catkin_ws/devel_isolated/gazebo_plugins/include/gazebo_plugins/GazeboRosCameraConfig.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:1655: CMakeFiles/gazebo_plugins_gencfg.dir/all] Error 2
make[1]: *** Attesa per i processi non terminati....
[ 33%] Linking CXX shared library /home/ale/catkin_ws/devel_isolated/gazebo_plugins/lib/libgazebo_ros_video.so
[ 33%] Built target gazebo_ros_video
[ 34%] Linking CXX shared library /home/ale/catkin_ws/devel_isolated/gazebo_plugins/lib/libgazebo_ros_harness.so
[ 34%] Built target gazebo_ros_harness
[ 35%] Linking CXX shared library /home/ale/catkin_ws/devel_isolated/gazebo_plugins/lib/libgazebo_ros_gpu_laser.so
[ 35%] Built target gazebo_ros_gpu_laser
make: *** [Makefile:139: all] Error 2
<== Failed to process package 'gazebo_plugins':
  Command '['/opt/ros/lunar/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/ale/catkin_ws/build_isolated/gazebo_plugins && /opt/ros/lunar/env.sh make -j4 -l4

Command failed, exiting.
```

j-rivero, 2017-06-26:
Looks like the building system is not able to find the dynamic_reconfigure python module. Some question:

 * Is the dynamic_reconfigure ROS package installed in the system?
 * Did you source the setup.bash script before the build? If so, what is the value of the $PYTHONPATH enviroment variable after sourcing it?
 * What is the result of running `find /opt/ros/lunar/lib/python2.7/dist-packages -name dynamic*`

Alessandro-Barbieri, 2017-06-26:
My full commandline

`sudo LDFLAGS="${LDFLAGS} ${OPTLDFLAGS}" CFLAGS="${CFLAGS} ${OPTCFLAGS}" CXXFLAGS="${CXXFLAGS} ${OPTCXXFLAGS}" PYTHONPATH="/opt/ros/lunar/lib64/python2.7/site-packages" CATKIN_PREFIX_PATH=/opt/ros/lunar ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/lunar -DSETUPTOOLS_DEB_LAYOUT=OFF -DBoost_PYTHON3_FOUND=1 -DBoost_PYTHON3_LIBRARY_RELEASE=/usr/lib64/libboost_python-3.5.so`

>Is the dynamic_reconfigure ROS package installed in the system?

It's my first install, how do I check?

>Did you source the setup.bash script before the build?

No because it's the first installation

>What is the result of running find /opt/ros/lunar/lib/python2.7/dist-packages -name dynamic*

Do you mean `/opt/ros/lunar/lib64/python2.7/site-packages` ?

```
find /opt/ros/lunar/lib64/python2.7/site-packages/ -name dynamic*
/opt/ros/lunar/lib64/python2.7/site-packages/genpy/dynamic.py
/opt/ros/lunar/lib64/python2.7/site-packages/genpy/dynamic.pyc
/opt/ros/lunar/lib64/python2.7/site-packages/dynamic_reconfigure
/opt/ros/lunar/lib64/python2.7/site-packages/dynamic_reconfigure-1.5.48-py2.7.egg-info
```

j-rivero, 2017-06-26:
> What is the result of running find /opt/ros/lunar/lib/python2.7/dist-packages -name dynamic*
>
> Do you mean /opt/ros/lunar/lib64/python2.7/site-packages ?
>
> find [...]

There are some differences across platforms but your PYTHONPATH command in the compilation line should have made python to find the dynamic_reconfigure module (the error is not point to this, so all makes sense until here). Let's check if you have the msg subdirectory under `/opt/ros/lunar/lib64/python2.7/site-packages/dynamic_reconfigure/msg/`? Does it exists in your system?

Alessandro-Barbieri, 2017-06-27:
>Let's check if you have the msg subdirectory under /opt/ros/lunar/lib64/python2.7/site-packages/dynamic_reconfigure/msg/? Does it exist in your system?

I don't have that folder

j-rivero, 2017-06-27:
> I don't have that folder

Then the error makes sense. In my Ubuntu system using .deb packages:

```
jrivero@nium ~ $ sudo dpkg -L ros-lunar-dynamic-reconfigure  | grep msg
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_GroupState.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_ConfigDescription.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_Config.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_IntParameter.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_ConfigDescription.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_Group.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_GroupState.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_ParamDescription.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_BoolParameter.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_DoubleParameter.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_SensorLevels.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/__init__.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_BoolParameter.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_ParamDescription.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/__init__.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_Group.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_StrParameter.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_IntParameter.pyc
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_Config.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_SensorLevels.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_StrParameter.py
/opt/ros/lunar/lib/python2.7/dist-packages/dynamic_reconfigure/msg/_DoubleParameter.py
...
```

You will need to investigate why your dynamic_reconfigure installation does not have (or have in a different path) the `msg` directory.

Alessandro-Barbieri, 2017-06-27:
Found the arcane, them were installed in `/opt/ros/lunar/lib/python2.7/site-packages/dynamic_reconfigure/msg/`

j-rivero, 2017-06-27:
Perfect. There is a divergence in the `lib` directories used inside the dynamic_reconfigure installation on Gentoo. I close the issue here, nothing to do here in `gazebo_ros_pkgs` .

closed 2017-06-27 by j-rivero