ros-simulation/gazebo_ros_pkgs #595: When deleteing a robot model with gazebo_ros_camera gives Gazebo Aborts

RDaneelOlivav, 2017-07-01:
When adding to a URDF the following code, and spawned all is OK.

```
<gazebo reference="camera_link">
    <sensor type="camera" name="camera1">
      <update_rate>10.0</update_rate>
      <camera name="head">

          <pose>0 0 0 0 0 1.57</pose>

        <horizontal_fov>1.3962634</horizontal_fov>
        <image>
          <width>400</width>
          <height>400</height>
          <format>R8G8B8</format>
        </image>
        <clip>
          <near>0.01</near>
          <far>100</far>
        </clip>
        <noise>
          <type>gaussian</type>
          <stddev>0.007</stddev>
        </noise>
      </camera>
      <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
        <alwaysOn>true</alwaysOn>
        <updateRate>0.0</updateRate>
        <cameraName>myrobot/camera1</cameraName>
        <imageTopicName>image_raw</imageTopicName>
        <cameraInfoTopicName>camera_info</cameraInfoTopicName>
        <frameName>camera_link</frameName>
        <hackBaseline>0.07</hackBaseline>
        <distortionK1>0.0</distortionK1>
        <distortionK2>0.0</distortionK2>
        <distortionK3>0.0</distortionK3>
        <distortionT1>0.0</distortionT1>
        <distortionT2>0.0</distortionT2>
      </plugin>
    </sensor>
  </gazebo>
```

The problem comes when you delete the model while gazebo is still running. Makes Gazebo stop without finishing.

> pure virtual method called
> terminate called without an active exception
> Aborted (core dumped)
> [gazebo-2] process has died [pid 14734, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/rdaneel/catkin_ws/src/launch_worlds/worlds/urdf_demo.world __name:=gazebo 

My theory is that maybe the plugin loses the link reffered to and breaks.
Is there a way to indicate the plugin to stop or wait or even unload it so that the model can be repsawned?

Thankyou

MulixBF, 2018-02-02:
I am struggling with this problem too. Was you able to find a workaround?

ironmig, 2018-05-29:
354 was fixed and closed. Do either of you still have this issue? I am unable to reproduce it