ros-simulation/gazebo_ros_pkgs #596: Gazebo8 stops publishing topic after `Shutdown` of subscriber

nordmoen, 2017-07-03:
I'm having a strange problem in Gazebo8, but not in Gazebo7. In my code I have a simulation loop which will interact with a robot for 10 seconds and then evaluate the behavior of the robot. This loop is performed for many different parameters and needs to be performed many (10 000+) times. To collect data for only the 10 seconds of the given parameters I subscribe to the topic `/gazebo/link_states` and unsubscribe later (by calling `shutdown`) when the 10 second evaluation has finished. In Gazebo7 (tested with both Ubuntu and Fedora) this worked perfectly and no problems where detected, but then I had an unrelated problem and needed to upgrade to Gazebo8 on Ubuntu which started to exhibit the bad behavior.

If I use `rostopic hz /gazebo/link_states` I can see it being output during the first loop, but then it stops being published. This happens for both `/gazebo/link_states` and `/gazebo/model_states` and only under Gazebo8.

Is there anything wrong with unsubscribing from a topic using `shutdown`? I'm unsure how else best to collect data for only the duration of the evaluation loop.

nordmoen, 2017-07-03:
I created a minimal example demonstrating my problem here: https://github.com/nordmoen/test_gazebo8/tree/master

The main code that does anything is:
```cpp
for(int i = 0; i < 5; ++i) {
  ROS_INFO("Starting iteration %i", i);
  auto sub = nh.subscribe("/gazebo/link_states", 5, callback);
  wait.sleep();
  sub.shutdown();
  ROS_INFO("Ending iteration %i", i);
}
```

Which result in these results:

In Gazebo 7 (tested with 7.4) the expected output is something like:
```bash
[ INFO] [1499083983.423210418, 0.203000000]: Gazebo started. Starting loop
[ INFO] [1499083983.423258111, 0.203000000]: Starting iteration 0
[ INFO] [1499083984.223547626, 1.000000000]: Got link_states!
[ INFO] [1499083985.226454890, 2.000000000]: Got link_states!
[ INFO] [1499083986.233047563, 3.000000000]: Got link_states!
[ INFO] [1499083987.238373358, 4.000000000]: Got link_states!
[ INFO] [1499083988.243084799, 5.000000000]: Got link_states!
[ INFO] [1499083988.460649478, 5.216000000]: Ending iteration 0
[ INFO] [1499083988.460723512, 5.216000000]: Starting iteration 1
[ INFO] [1499083989.246983436, 6.000000000]: Got link_states!
[ INFO] [1499083990.251303354, 7.000000000]: Got link_states!
[ INFO] [1499083991.266260549, 8.000000000]: Got link_states!
[ INFO] [1499083992.271432231, 9.000000000]: Got link_states!
[ INFO] [1499083993.276270854, 10.000000000]: Got link_states!
...
```

In Gazebo 8 (tested with 8.1.1) the output is instead:
```bash
[ INFO] [1499082452.365551220, 0.247000000]: Gazebo started. Starting loop
[ INFO] [1499082452.365563268, 0.247000000]: Starting iteration 0
[ INFO] [1499082453.119217355, 1.000000000]: Got link_states!
[ INFO] [1499082454.120610593, 2.000000000]: Got link_states!
[ INFO] [1499082455.123890395, 3.000000000]: Got link_states!
[ INFO] [1499082456.221480809, 4.001000000]: Got link_states!
[ INFO] [1499082457.217972288, 5.001000000]: Got link_states!
[ INFO] [1499082457.465672239, 5.252000000]: Ending iteration 0
[ INFO] [1499082457.465704627, 5.252000000]: Starting iteration 1
[ INFO] [1499082462.783421969, 10.263000000]: Ending iteration 1
[ INFO] [1499082462.783455855, 10.263000000]: Starting iteration 2
[ INFO] [1499082468.489491249, 15.299000000]: Ending iteration 2
[ INFO] [1499082468.489523642, 15.299000000]: Starting iteration 3
[ INFO] [1499082473.496997661, 20.323000000]: Ending iteration 3
[ INFO] [1499082473.497029367, 20.323000000]: Starting iteration 4
[ INFO] [1499082478.510929082, 25.328000000]: Ending iteration 4
[ INFO] [1499082478.510962179, 25.328000000]: Ending 'test_gazebo8_node'

```

batman177, 2017-09-21:
+1, same issue here when using rostopic echo -n1 multiple times on /gazebo/model_states

lcapkovic, 2018-01-09:
+1, I'll add that `/gazebo/link_states` is also affected by this, but they die independently. Running this:
```
rostopic echo -n1 /gazebo/model_states
rostopic echo -n1 /gazebo/link_states
```
gets a message for both, but subsequent calls of the same fail.

jbrookshire, 2018-01-11:
+1.  Running the following alongside Gazebo helped me work around the issue in the meantime:

`rosrun topic_tools relay /gazebo/link_states /link_states`

and then subscribe to /link_states instead.

nordmoen, 2018-02-02:
I just updated my source install, `gazebo_ros_pkgs` version `2.7.2`, and the problem has vanished. Unfortunately I have not noted the version I upgraded from so I can't tell which commit fixed the issue, but running `test_gazebo8` now shows the same behaviour as Gazebo7.

If anyone else can comment as to if the problem is fixed for them and maybe even which version it did and didn't work on we can close this issue.

Brudasoe, 2018-02-13:
I'm having the same problem, the messages sent by, for example force_torque sensor or joint_states topics seem to stop publishing for an increasingly longer amount of time after sim reset, I tried writing a workaround that waits for them to start publishing, and I suspect that they are waiting for some internal counter (maybe seqence ID in the header?) to go to the value it was before the reset. I tried reinstalling `gazebo_ros_pkgs`, but this didn't work for me (maybe I did something incorrectly).

eborghi10, 2018-04-25:
@nordmoen It seems that you were using ROS Kinetic, right? Because I'm having the same issue with ROS Lunar and Gazebo 8. In my case, I have `gazebo8_ros_pkgs` version `2.7.2`.

nordmoen, 2018-04-26:
@eborghi10 I'm using Lunar as well. Recently upgraded to Gazebo `8.1.1` so maybe there is some interaction between `gazebo_ros_pkgs` and Gazebo causing this.

eborghi10, 2018-04-26:
I'm using Gazebo `8.4.0` but I found yesterday that I cannot subscribe again to the `/gazebo/model_states` 
 topic (as an example) once I call to [`unregister()`](http://docs.ros.org/lunar/api/rospy/html/rospy.topics.Topic-class.html#unregister).

ironmig, 2018-05-25:
Here is the testing I have done on this issue:
gazebo 8.4.0 with gazebo_ros_pkgs 2.7.2 -> fails
gazebo 8.4.0 with gazebo_ros_pkgs 2.7.3 -> passes
gazebo 8.4.0 with gazebo_ros_pkgs 2.7.2 AGAIN -> fails again