ros-simulation/gazebo_ros_pkgs #597: spawn_model logging issue

joao-alex-cunha, 2017-07-12:
I was trying to spawn a model in gazebo at runtime and I misread the proper usage message from spawn_model script. Not surprisingly the model wasn't spawning but I could understand why since the logging messages weren't appearing. Instead you get:

```
SpawnModel script started
No handlers could be found for logger "rosout"
```

and then a silent exit. On further inspection I found out that the messages weren't appearing since `rospy.init_node('spawn_model')` is invoked **after** the parsing of the user inputs, so these messages get dropped.

A solution would be to change all messages in the parsing phase to python prints or another logging framework, or to actually invoke `rospy.init_node('spawn_model')` before parsing the CLI inputs.

To reproduce this issue in a MWE open a terminal and start gazebo:
```
roslaunch gazebo_ros empty_world.launch
```
and in another terminal write:
```
rosrun gazebo_ros spawn_model -sdf -name cricket_ball -database cricket_ball
```
(the -name option doesn't exist, to make it work replace with -model). You might need to have the cricket_ball model in your local gazebo models folder.

I'm using ROS kinetic and Ubuntu 16.04

pradhannihar, 2017-12-28:
i got the same issue but the exact way to spwan a model is

_`rosrun gazebo spawn_model -file `pwd`/object.urdf -urdf -z 1 -model my_object`_

In my command the  -urdf was missing

ironmig, 2018-07-06:
Fixed in #730

PR #730 (2018-07-06): Refactor spawn_model script
Implement #729:
*    parse arguments using argparse (much less code and easier to read than manually parsing arguments)
*    Fix #597 (argument parsing errors are printed instead of roslogged as they occur before node initialization)
*    Remove deprecated -gazebo option
*    Remove references to trimesh format and options (they were never parsed in the script)
*    Exit with non-zero exit code on failure so scripts can see if spawn_model succeeded
*    Initialize node anonymously so multiple instances can run simultaneously
*    Various style (pep8 compliant) / rea [... cut]
closed 2018-07-06 by ironmig