ros-simulation/gazebo_ros_pkgs #608: Does gazebo_ros_pkgs support bullet now?

longwoo, 2017-08-17:
I am trying to use bullet engine in Gazebo8+Ros kinetic, since ODE engine doesn't support rolling friction.

so I use args like this in launch file
```
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
...
    <arg name="physics" value="bullet" /> 
  </include>
```

however, it gives me 

``` [ERROR] [1502890717.457463220, 0.206000000]: ROS get_physics_properties service call does not yet support physics engine [bullet].```

The gazebo started anyway and with the GUI shown the engine is BULLET.

Second,in the left side of Gazebo GUI, the Property doesn't match SDF1.6 specification ,for example, the item collision->surface->friction .

j-rivero, 2017-08-17:
> The gazebo started anyway and with the GUI shown the engine is BULLET.

This seems to me that the launch file passes the `-e` argument successfully to Gazebo, so yes, the simulator is launched using Bullet.

> [ERROR] [1502890717.457463220, 0.206000000]: ROS get_physics_properties service call does not yet support physics engine [bullet].

I'm pretty sure [looking at the code](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L1322) that `gazebo_ros_pkgs` does not have the proper support for other physics engines than ode. It should not be difficult to implement, if someone want to code it I will happily review it.

> Second,in the left side of Gazebo GUI, the Property doesn't match SDF1.6 specification ,for example, the item collision->surface->friction .

It is the problem also present when using Gazebo without ROS? If so, there is a bug in upstream Gazebo. If the problem is not present when using standalone Gazebo, then it is probably a consequence of the lack of support in gazebo_ros_pgks for other physics engines that we described in the sentence above.

longwoo, 2017-08-18:
@j-rivero Thanks. The   `Property doesn't match SDF1.6 specification` problem still exsit if I start Gazebo with cmd ```gazebo -e bullet``, I will report it to gazebo forum.

j-rivero, 2017-08-21:
> @j-rivero Thanks. The Property doesn't match SDF1.6 specification problem still exsit if I start Gazebo with cmd gazebo -e bullet, I will report it to gazebo forum.

Thanks very much. Looks like a bug more than question, you can consider to use the [gazebo issue tracker](https://bitbucket.org/osrf/gazebo/issues). I will close the issue by now, feel free to report the result or link from gazebo upstream or open it again if you found that we need to fix it from this repo.

closed 2017-08-21 by j-rivero