ros-simulation/gazebo_ros_pkgs #610: ros-indigo-gazebo-ros-control : position error since update

KevinKDufour, 2017-08-17:
Hello,

Since the august update of the ros-indigo-gazebo-ros-control package (2.4.15), I noticed some position errors and large oscillations in Gazebo with the Baxter Robot that weren't here before.

The simplest way I found to reproduce the problem is the following:

With the right arm, from the joint position [0,0,0,0,0,0,0] I sent the command 
`rostopic pub /robot/limb/right/joint_command baxter_core_msgs/JointCommand "{mode: 1, command: [-0.690766,-0.817243,-0.337815,0.699642,0.600842,1.13207,-0.670463], names: ['right_s0', 'right_s1', 'right_e0', 'right_e1', 'right_w0', 'right_w1', 'right_w2']}" -r 100`

The expected pose is shown in the file attached "ExpectedPose.png".

However, with the last ros-indigo-gazebo-ros-control version installed I get the wrong position. The video attached in "UpdatedPackage.mp4.zip" shows the oscillations and the wrong finale pose.

To solve this problem, I removed the ros-indigo-gazebo-ros-control package and installed the 2.4.13 version from source. After compilation and restarting Gazebo, I get the right pose, see the video attached in "PreviousPackage.mp4.zip".

I don't know if the ros-indigo-gazebo-ros-control package is really the problem but downgrading it solved the issue.

[image]

[PreviousPackage.mp4.zip](https://github.com/ros-simulation/gazebo_ros_pkgs/files/1232270/PreviousPackage.mp4.zip)
[UpdatedPackage.mp4.zip](https://github.com/ros-simulation/gazebo_ros_pkgs/files/1232269/UpdatedPackage.mp4.zip)

j-rivero, 2017-08-17:
> I don't know if the ros-indigo-gazebo-ros-control package is really the problem but downgrading it solved the issue.

Looks like we have a candidate, yes. It is weird that there is [no change from the 2.4.14 to 2.4.15](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/indigo-devel/gazebo_ros_control/CHANGELOG.rst). Is the 2.4.14 the version working for you? Could it be that the problem is in other gazebo_ros_pkgs package?

One option could be to perform a source installation inside a catkin ws (gazebo_ros_pkgs + baxter) and use the git bisect to detect which commit is the culprit. Sorry I can not provide more help.

KevinKDufour, 2017-08-18:
To give more details, the error appeared with the package installed using `sudo apt-get install ros-indigo-gazebo-ros-control`, which is version 2.4.15-0trusty-20170809-103121-0800.
To downgrade the version, as I wasn't able to do it with apt-get, I simply removed the package with apt-get and installed the 2.4.13 from source in my catkin ws. 

I tried to check the repeatability of the problem with version 2.4.15 (installed with apt-get or installed from source in catkin ws) and what I see is that: when I reset my catkin ws (by deleting build and devel folder and compiling again with catkin_make) and try to reproduce the problem and do these steps several times, sometimes it works, sometimes it doesn't. 
So this is very confusing as the package seems not to give the same results each time I reset my catkin ws.... Same repeatability problem appeared with 2.4.14 version, but not with 2.4.13 however. So i'm not sure what the problem is...

j-rivero, 2017-08-21:
> So this is very confusing as the package seems not to give the same results each time I reset my catkin ws.... Same repeatability problem appeared with 2.4.14 version, but not with 2.4.13 however. So i'm not sure what the problem is...

So the problem seems to be between 2.4.13 -> 2.4.14 which according to the Changelog there is no change in `gazebo_ros_control` package. Since you are using a catkin workspace, not only `gazebo_ros_control` version is changing, you are probably changing all `gazebo_ros_pkgs`. And the problem is not always reproducible, that is perfectly possible for example in the presence of a race condition. Again , I would recommend to go with a [bisect git method](https://git-scm.com/docs/git-bisect) for trying to detect which commit is exactly the culprit.

Thanks.

KevinKDufour, 2017-09-11:
Hi, sorry for the delay of answer, I wasn't available lately.

So I did more tests today, and I was able to reproduce the issue with version 2.4.13 as well: when I reset my catkin ws and compile several times, sometimes it works correctly, sometimes not.

I am only changing gazebo_ros_control and not gazebo_ros_pkgs. Indeed, to try other versions of gazebo_ros_pkgs, I would have to uninstall the current version installed via apt-get but I can't as apt-get wants to remove ros-indigo-desktop-full when I ask for removing gazebo_ros_pkgs.

Would you have any suggestion of test I could do to identify the responsible package and/or version?

Thanks