ros-simulation/gazebo_ros_pkgs #612: [gazebo_ros_control] Strange behaviour when adding gazebo_ros_control to a robot

Zabot, 2017-08-24:
I was modeling up an arm, and everything seemed to be working until I added the ros control plugin:
```xml
  <gazebo>
    <plugin filename="libgazebo_ros_control.so" name="gazebo_ros_control">
      <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
    </plugin>
  </gazebo>
```

The only difference between these two arms is that the arm in front has gazebo_ros_control plugin enabled, and the one in back has those lines commented out.

[image]

I've verified the inertial properties of are all set right, tried it with collisions disabled completely and without, its the same both ways. I am still able to control the joints, and they behave as expected, with normal speed, though sluggish acceleration.

Even more curious, when I manually change any one of the four revolute joints on the gazebo_ros_control arm to a fixed joint it behaves perfectly fine.

I was observing a similar issue with turtlebot_arm that I had originally attributed to inertial properties, but it seems like it may be a deeper seated issue.
turtlebot/turtlebot_arm/issues/22

Zabot, 2017-09-01:
After further investigation this appears to be related to #479 and specifically [this PR](https://bitbucket.org/osrf/gazebo/pull-requests/2598/fix2111/diff) to gazebo.

It looks like the function that was being used before to change the rotation of a frame used to preserve the velocity of the frame, and now it zeros the velocity of the frame. As such every time the position of a PositionJointInterface is updated, the velocity of the child frame is zeroed, preventing it from accelerating properly.

mintar, 2018-01-23:
Thanks @Zabot for writing this up so concisely. I just got bitten by this issue myself. I was working on a robot URDF consisting of a diffdrive mobile base with a robot arm on top. This issue caused the robot arm to act like a giant parachute, counteracting gravity, and causing some of the wheels to lose contact with the ground. It took me ages to figure out that an incorrect simulation of the arm was causing my diffdrive controller to fail...

Anyhow, for posterity here are the workarounds mentioned in ros-simulation/gazebo_ros_pkgs#479 :

* Add `pid_gains` for your Gazebo controllers when using the `PositionJointInterface`, like this:

```yaml
gazebo_ros_control:
  pid_gains:
    joint2: {p: 100.0, i: 0.01, d: 10.0}
```

* Alternatively, use the `VelocityJointInterface` or `EffortJointInterface` instead.

Both successfully work around this issue.

vincentrou, 2018-03-01:
This issue can be closed since there is workaround and it now works with gazebo9.

Zabot, 2018-03-01:
The PR over on gazebo was merged, but don't we still need to change `ros_control_plugin` to use the new flag that was added? If I'm not mistaken wasn't it decided that the default behavior would remain unchanged?

mintar, 2018-03-05:
Yes, please don't close this issue. It's still there in ROS Kinetic + Gazebo7, which is what most users use. IMO, if this bug won't get fixed, it would be nice if at least a big fat warning would be printed. Usually this bug is very hard to track down, because it manifests in unintuitive problems like the mobile base isn't moving any more once you add an arm to the robot (because this bug causes the mobile base to lose contact with the ground). 

There seems to be a large number of users affected - basically everyone who's using ros_control + Gazebo together. Every week, a couple of new questions pop up on ROS answers.

mintar, 2018-03-05:
> The PR over on gazebo was merged, but don't we still need to change ros_control_plugin to use the new flag that was added? If I'm not mistaken wasn't it decided that the default behavior would remain unchanged?

I agree. The PR [#2814](https://bitbucket.org/osrf/gazebo/pull-requests/2814/fix-issue-2111-by-providing-options-to/diff) just added the option to specify `preserveWorldVelocity`, but the default is `false`. The `ros_control_plugin` still has to set it to `true` in the call to `Joint::setPosition()`. Unfortunately, that PR has only been merged into Gazebo9, not Gazebo7.

mintar, 2018-03-06:
In case somebody is interested, I've written a tutorial package that shows how to work around this bug:

https://github.com/mintar/mimic_joint_gazebo_tutorial

j-rivero, 2018-03-09:
Thanks for all the debugging and information. If we need to backport a change from gazebo9 to previous versions, let me know. If we need to made changes in this repository, would you mind to create a PR with the needed changes?

mintar, 2018-03-16:
@j-rivero : I've added a pull request (#688) that fixes this issue for Gazebo 9. Could you please backport [gazebo PR 2814](https://bitbucket.org/osrf/gazebo/pull-requests/2814/fix-issue-2111-by-providing-options-to/diff) to Gazebo 7 and Gazebo 8, so that we can relax the GAZEBO_MAJOR_VERSION guards in #688 to bring this fix to Gazebo 7?

scpeters, 2018-03-16:
I'm going to see if it's possible to backport that bugfix to gazebo7 without breaking API/ABI

I'm testing a branch right now

scpeters, 2018-03-16:
[image] https://build.osrfoundation.org/job/gazebo-abichecker-any_to_any-xenial-amd64/571

scpeters, 2018-03-17:
I think we had talked about this at the time, but we couldn't figure out a way to backport a fix without breaking API/ABI

tried tweaking the `const` qualifiers, but that didn't help

[image] https://build.osrfoundation.org/job/gazebo-abichecker-any_to_any-xenial-amd64/572

https://bitbucket.org/osrf/gazebo/branches/compare/issue_2111_7%0Dgazebo7#diff

closed 2018-03-17 by scpeters

mintar, 2018-03-19:
@scpeters : Wouldn't something like this ABI compatible?

```cpp
bool Joint::SetPosition(unsigned int _index, double _position)
{
   return this->SetPosition(_index, _position, false);
}

bool Joint::SetPosition(const unsigned int _index, const double _position,
                        const bool _preserveWorldVelocity)
{
  // ...
}
```

That way, you wouldn't be removing any symbols, only adding new ones. It would be really nice if we could backport this to Gazebo 7. Otherwise we should add a big fat warning whenever people call SetPosition without preserveWorldVelocity, since it will break their simulation in unexpected and hard to debug ways.

TheDash, 2018-06-11:
I'm also having this issue... except the pid_gains is *not* fixing it =(

Any other suggestions?

mintar, 2018-06-11:
I've added a warning message in #691 that should tell you if you're affected by exactly this issue or not. Unfortunately, that warning is only part of release 2.5.16, and kinetic is still on 2.5.14, so you'll have to build gazebo_ros_pkgs from source to get that warning.

j-rivero, 2018-06-11:
> I've added a warning message in #691 that should tell you if you're affected by exactly this issue or not. Unfortunately, that warning is only part of release 2.5.16, and kinetic is still on 2.5.14, so you'll have to build gazebo_ros_pkgs from source to get that warning.

Waiting next sync in the shadow fixed repo: http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-kinetic-gazebo-ros-pkgs/ in the case you want to use .deb packages. Thanks @mintar for the contribution.

PR #688 (2018-03-17): Fix #612 for Gazebo9
This commit fixes #612, but only for Gazebo9. Fixing it for Gazebo7
(the version used in ROS Kinetic) requires the following PR to be
backported to Gazebo 7 and 8:

https://bitbucket.org/osrf/gazebo/pull-requests/2814/fix-issue-2111-by-providing-options-to/diff

Once that PR has been backported, we can remove the GAZEBO_MAJOR_VERSION
guards from this PR so that the fix is active for the older Gazebo
versions as well.

Tested on Gazebo7 (where it compiles, but doesn't change anything) and
Gazebo9 (where it compiles and fixes the bug). I've tested it using the
instructions I [... cut]
commit 3164e4c6 (2018-03-17): Fix #612 for Gazebo9 (#688)

* Fix #612 for Gazebo9

This commit fixes #612, but only for Gazebo9. Fixing it for Gazebo7
(the version used in ROS Kinetic) requires the following PR to be
backported to Gazebo 7 and 8:

https://bitbucket.org/osrf/gazebo/pull-requests/2814/fix-issue-2111-by-providing-options-to/diff

Once that PR has been backported, we can remove the GAZEBO_MAJOR_VERSION
guards from this PR so that the fix is active for the older Gazebo
versions as well.

Tested on Gazebo7 (where it compiles, but doesn't change anything) and
Gazebo9 (where it compiles and fixes the bug). I've te [... cut]
PR #699 (2018-04-05): Fix #612 for Gazebo9 (lunar-devel)
{ port of pull request #688 }
This commit fixes #612, but only for Gazebo9. Fixing it for Gazebo7
(the version used in ROS Kinetic) requires the following PR to be
backported to Gazebo 7 and 8:

https://bitbucket.org/osrf/gazebo/pull-requests/2814/fix-issue-2111-by-providing-options-to/diff

Once that PR has been backported, we can remove the GAZEBO_MAJOR_VERSION
guards from this PR so that the fix is active for the older Gazebo
versions as well.

Tested on Gazebo7 (where it compiles, but doesn't change anything) and
Gazebo9 (where it compiles and fixes the b [... cut]
commit cb997c49 (2018-04-05): Fix #612 for Gazebo9 (lunar-devel) (#699)

* Fix #612 for Gazebo9

This commit fixes #612, but only for Gazebo9. Fixing it for Gazebo7 (the version used in ROS Kinetic) requires the following PR to be backported to Gazebo 7 and 8: