ros-simulation/gazebo_ros_pkgs #614: [gazebo_ros] ROS not initialized in gzclient for a GUIPlugin

romainreignier, 2017-10-13:
I have tried to implement a [GUIPlugin](http://gazebosim.org/tutorials?tut=gui_overlay) for Gazebo, using ROS to subscribe to a topic but ROS is not initialized.

In fact, GUIPlugin is used in `gzclient` and not `gzserver` like the others `ModelPlugin` and `SensorPlugin`, so there is no ROS node in gzclient.

As a quick workaround, I have called `ros::init()` in the `Load()` method of my plugin.

But I think it could be useful to add a ROS node mechanism in the `GazeboRosPathsPlugin`.

prateek470, 2018-05-09:
I am facing the same problem. Also after ros::init in one GUIPlugin, other GUIPlugins stops working. Please tell me if there is any solution. I am using Gazebo8 + ROS Kinetic.

iche033, 2018-05-09:
I think this is fixed by pull request #676 for lunar and pull request #670 for kinect. What ROS version are you using?

iche033, 2018-05-10:
Actually looks like the changes have not been released yet. You'll need to install gazebo_ros_pkgs from source to get the fix. Alternatively, looking at the changes in the fix, you could probably just manually load `libgazebo_ros_api_plugin.so` when launching gzclient

prateek470, 2018-05-10:
Thanks, @iche033 Will test and update here soon.

ironmig, 2018-06-15:
@prateek470 Did you ever get a chance to test with those new patches? Nice to close the issue if it is fixed

closed 2018-06-15 by ironmig

romainreignier, 2018-06-18:
Indeed it works now with the git version (not yet applied in the gazebo8 package on Kinetic).

Thanks