ros-simulation/gazebo_ros_pkgs #615: Where does code exist

Kukanani, 2017-10-17:
Does the indigo-devel branch of this repository target Gazebo 2 or Gazebo 7? If it targets 2, where (what branch?) is the Gazebo 7 version? I'm trying to make extensions to the `ingido-devel` branch targeting gz7, but I'm getting errors that suggest that this branch targets gz, of the type:
``
[Err] [ODEPhysics.cc:1534] ODEPhysics::SetParam(precon_iters) boost::any_cast error: boost::bad_any_cast: failed conversion using boost::any_cast
```
I wasn't able to find this information readily available elsewhere.

Kukanani, 2017-10-17:
My question still stands, however, after looking into my errors further it appears that they occur in Gazebo7/Kinetic configuration as well. Opening a new ticket for that.

j-rivero, 2017-12-14:
There are official gazebo versions supported by ROS, for Indigo the official version is gazebo 2. However we create some 'unofficial' gazebo_ros_pkgs packages using modern gazebo versions (more information http://gazebosim.org/tutorials/?tut=ros_wrapper_versions).

So: the Indigo branch should work with both, gazebo2 and gazebo7. The Kinetic branch should work with the official supported version, gazebo7, but also with gazebo8.

I hope that this answers your questions. Feel free to reopen if it does not.

closed 2017-12-14 by j-rivero