ros-simulation/gazebo_ros_pkgs #616: Boost bad_any_cast while setting physics properties via gazebo_ros_pkgs

Kukanani, 2017-10-17:
I am receiving the following errors when attempting to set physics properties via the ROS API. The errors are thrown in Gazebo, so it's possible that this is a Gazebo issue, but I thought I would post it here since my current reproducible example uses the ROS API. The errors are:
```
[Err] [ODEPhysics.cc:1534] ODEPhysics::SetParam(precon_iters) boost::any_cast error: boost::bad_any_cast: failed conversion using boost::any_cast
[Err] [ODEPhysics.cc:1534] ODEPhysics::SetParam(iters) boost::any_cast error: boost::bad_any_cast: failed conversion using boost::any_cast
[Err] [ODEPhysics.cc:1534] ODEPhysics::SetParam(max_contacts) boost::any_cast error: boost::bad_any_cast: failed conversion using boost::any_cast
```

Here is a minimal test case, in which I get the physics parameters first so that I can set the solver parameters correctly in the setter message:
```
import rospy
import time

from gazebo_msgs.msg import *
from gazebo_msgs.srv import *
from std_srvs.srv import Empty
from geometry_msgs.msg import Point, Pose, Quaternion, Twist, Vector3

from xml.etree.ElementTree import Element, SubElement, tostring

rospy.init_node('test')
ns = "gazebo"
rospy.wait_for_service(ns + '/get_physics_properties')
rospy.wait_for_service(ns + '/set_physics_properties')

try:
    get_physics_properties = \
        rospy.ServiceProxy(ns + '/get_physics_properties',
                           GetPhysicsProperties)
    props = get_physics_properties()
except rospy.ServiceException as e:
    rospy.logerr('couldn\'t get physics properties while preparing '
                 'to set physics properties: ' + str(e))

time_step = props.time_step
max_update_rate = props.max_update_rate
gravity = props.gravity

try:
    set_physics_properties = \
        rospy.ServiceProxy(ns + '/set_physics_properties',
                           SetPhysicsProperties)
    set_physics_properties(time_step=time_step,
                           max_update_rate=max_update_rate,
                           gravity=gravity,
                           ode_config=props.ode_config)
except rospy.ServiceException as e:
    rospy.logerr('couldn\'t set physics properties: ' + str(e))
```

ironmig, 2018-05-25:
Could not replicate running this script on my system. Please give more details on what you're running:
* gazebo version
* ROS version
* Ubuntu version
* Gazebo world

Kukanani, 2018-05-29:
Hi Kevin, thanks for pinging back on this. Unfortunately I encountered this bug quite some time ago and no longer have the details (I've moved on to other research projects). I expect that this would have been Ubuntu 16.04, ROS Kinetic, and Gazebo 7.x.x since that's the system I was using months ago. However, I can't provide any more details, so unless this clearly is still a bug with the system specs I described above, it's probably best to close the issue. Thanks!

j-rivero, 2018-05-29:
Thanks @ironmig, @Kukanani . Please comment if we need to reopen the issue.

closed 2018-05-29 by j-rivero