ros-simulation/gazebo_ros_pkgs #643: [gazebo_ros] gzserver script overwrites GAZEBO_MODEL_DATABASE_URI

rsmallegoor, 2017-12-05:
When using Gazebo without ROS, it is possible to set a custom GAZEBO_MODEL_DATABASE_URI. This is required to load models from a specif model server instead of the [default server](http://gazebosim.org/models).

Unfortunately, when using `gazebo_ros`, this variable is overwritten [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/0b99225ae4c3f098a8cbc3873ac312376dd9a1be/gazebo_ros/scripts/gzserver#L30). Is there anything that can be done to prevent this?

If not, should the [script](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/0b99225ae4c3f098a8cbc3873ac312376dd9a1be/gazebo_ros/scripts/gzserver) be modified to handle the GAZEBO_MODEL_DATABASE_URI the same way as the GAZEBO_MASTER_URI variable?

rsmallegoor, 2017-12-05:
[PR](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/644)

j-rivero, 2017-12-05:
I think that we can fix the problem with the same solution applied to the `GAZEBO_MASTER_URI` as you suggested.

Another interesting experiment would be to remove completely the call to the setup.sh. Gazebo build system set [default values for all the variable if they are not defined](https://bitbucket.org/osrf/gazebo/src/b27e852937eec3b5f1f3829cc4535b42c1c66a00/gazebo/common/CMakeLists.txt?at=default&fileviewer=file-view-default#CMakeLists.txt-218).

rsmallegoor, 2017-12-05:
#644

j-rivero, 2017-12-15:
Merged and ported to lunar-devel. Thanks for the PR.

closed 2017-12-15 by j-rivero