ros-simulation/gazebo_ros_pkgs #645: differential_drive_controller plugin causes robot rotation with no cmd_vel input

pirobot, 2017-12-10:
This happens in both Indigo and Kinetic (Gazebo 2.3 and 7.0).    Following [this tutorial](http://gazebosim.org/tutorials?tut=build_robot) to create a simple differential drive robot and then bringing the robot up in Gazebo, I discovered that the robot rotates (apparently always counterclockwise) at a slow but significant rate even without cmd_vel input.  The robot also has non-zero linear x and y velocities.

I actually discovered this issue with a custom robot model that uses the differential_drive_controller plugin under ROS Indigo so I tried it with this simpler robot under both Indigo and Kinetic.  I list my SDF file below for reference.  The only thing I added was some friction to the drive wheels to see if that fixed the problem but it does not.

Unless I am missing an important parameter, this seems like a fairly serious bug since a long running Gazebo simulation of a differential drive robot will suffer from a large drift in the robot's position.  (So I hope I'm just doing something wrong!)

Thanks!
patrick

my_robot.sdf
```
<?xml version='1.0'?>
<sdf version='1.4'>
  <model name="my_robot">
     <static>false</static>
     <link name='chassis'>
        <pose>0 0 0.1 0 0 0</pose>
        <inertial>
           <mass>10.0</mass>
           <pose>0.0 0 0.1 0 0 0</pose>
           <inertia>
              <ixx>0.5</ixx>
              <iyy>1.0</iyy>
              <izz>0.1</izz>
           </inertia>
        </inertial>
        <collision name='collision'>
            <geometry>
                <box>
                   <size>0.4 0.2 0.1</size>
                </box>
            </geometry>
        </collision>
        <visual name='visual'>
           <pose>0 0 0.0 0 0 0</pose>
           <geometry>
              <box>
                 <size>0.4 0.2 0.1</size>
              </box>
           </geometry>
         </visual>
         <collision name='caster_collision'>
            <pose>-0.15 0 -0.05 0 0 0</pose>
            <geometry>
               <sphere>
                  <radius>.05</radius>
               </sphere>
            </geometry>
            <surface>
              <friction>
                 <ode>
                    <mu>0</mu>
                    <mu2>0</mu2>
                    <slip1>1.0</slip1>
                    <slip2>1.0</slip2>
                 </ode>
              </friction>
           </surface>
        </collision>
        <visual name='caster_visual'>
           <pose>-0.15 0 -0.05 0 0 0</pose>
           <geometry>
              <sphere>
                 <radius>.05</radius>
              </sphere>
           </geometry>
         </visual>
     </link>

     <link name="right_wheel">
        <pose>0.1 0.13 0.1 0 1.5707 1.5707</pose>
        <inertial>
           <pose>0.1 0.13 0.1 0 1.5707 1.5707</pose>
           <mass>1.0</mass>
           <inertia>
              <ixx>0.0045</ixx>
              <iyy>0.005</iyy>
              <izz>0.0045</izz>
           </inertia>
        </inertial>
        <collision name="collision">
          <geometry>
              <cylinder>
                 <radius>.1</radius>
                 <length>.05</length>
              </cylinder>
          </geometry>
            <surface>
              <friction>
                 <ode>
                    <mu>10</mu>
                    <mu2>10</mu2>
                 </ode>
              </friction>
           </surface>
        </collision>
        <visual name="visual">
          <geometry>
              <cylinder>
                 <radius>.1</radius>
                 <length>.05</length>
              </cylinder>
          </geometry>
        </visual>
      </link>

      <link name="left_wheel">
        <pose>0.1 -0.13 0.1 0 1.5707 1.5707</pose>
        <inertial>
           <pose>0.1 0.13 0.1 0 1.5707 1.5707</pose>
           <mass>1.0</mass>
           <inertia>
              <ixx>0.0045</ixx>
              <iyy>0.005</iyy>
              <izz>0.0045</izz>
           </inertia>
        </inertial>
        <collision name="collision">
           <geometry>
              <cylinder>
                 <radius>0.1</radius>
                 <length>0.05</length>
              </cylinder>
           </geometry>
            <surface>
              <friction>
                 <ode>
                    <mu>10</mu>
                    <mu2>10</mu2>
                 </ode>
              </friction>
           </surface>
        </collision>
        <visual name="visual">
           <geometry>
              <cylinder>
                 <radius>0.1</radius>
                 <length>0.05</length>
             </cylinder>
           </geometry>
        </visual>
      </link>

      <joint type="revolute" name="left_wheel_joint">
         <pose>0 0 -0.03 0 0 0</pose>
         <child>left_wheel</child>
         <parent>chassis</parent>
         <axis>
            <xyz>0 1 0</xyz>
         </axis>
      </joint>

      <joint type="revolute" name="right_wheel_joint">
         <pose>0 0 0.03 0 0 0</pose>
         <child>right_wheel</child>
         <parent>chassis</parent>
         <axis>
            <xyz>0 1 0</xyz>
         </axis>
      </joint>

      <plugin name="differential_drive_controller" filename="libgazebo_ros_diff_drive.so">
         <alwaysOn>true</alwaysOn>
         <updateRate>20</updateRate>
         <leftJoint>left_wheel_joint</leftJoint>
         <rightJoint>right_wheel_joint</rightJoint>
         <wheelSeparation>0.4</wheelSeparation>
         <wheelDiameter>0.1</wheelDiameter>
         <wheelTorque>20</wheelTorque>
         <!--wheelAcceleration>5</wheelAcceleration-->
         <commandTopic>cmd_vel</commandTopic>
         <odometryTopic>odom</odometryTopic>
         <odometryFrame>odom</odometryFrame>
         <publishTf>1</publishTf>
         <publishWheelJointState>true</publishWheelJointState>
         <robotBaseFrame>base_footprint</robotBaseFrame>
      </plugin>

   </model>
</sdf>

```

howardcochran, 2018-01-24:
I'd just like to mention that we also suffer from this problem using a simple differential drive robot model.

j-rivero, 2018-01-25:
Could you please give a try using the latest gazebo7 release (7.9.0) from the osrfoundation package repository: http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install

pirobot, 2018-01-29:
I just gave gazebo7 release 7.9.0 a try with the above model under ROS Kinetic and I still see the same problem although I believe the behavior has changed a little bit as follows:
- when I first launch Gazebo with the robot model, there is a very slight counter-clockwise rotation of the robot which then stops after about 30 seconds.
- if I start rotating the robot by publishing a Twist message on the /cmd_vel topic and then try to stop the robot by continually publishing an empty Twist message on /cmd_vel, the robot continues to rotate with a small angular velocity.

ironmig, 2018-05-24:
I briefly tried debugging this. I found the plugin to set joint velocities to 0 every loop still caused some motion. With no controller running, the model moved around in the simulator also, rolling forward and veering to one side. These two points lead me to believe it is not a bug in the diff drive plugin. I think this is likely just a consequence of the model or gazebo's physics simulation of it, which makes it hard to control with such a simple controller.

pirobot, 2018-09-02:
I just came back to this after several months of using the ROS diff_drive_controller instead of the plugin.  So I tried the plugin again with Gazebo release 7.14, Unbuntu 16.04 and ROS Kinetic and the spontaneous rotation seems to have gone away.

closed 2018-09-02 by pirobot

grafoteka, 2018-10-26:
Hi all! I'm launching the robot Summit XL in Gazebo and RViz, and without giving it any move command (linear or angular) it turns.

I have update to Gazebo 7.14 with Xubuntu 16.04 but the problem still been there. 

The robot is using the Diff_drive plugin.

grafoteka, 2018-10-26:
I have just upload a video of what is happening

[Video](https://youtu.be/qQ0mNyko1Tw)

I try to remove the Gazebo plugin, but the error persists.

@chapulina  I will ask there. Thank you

wendwosen, 2019-06-20:
@grafoteka Did you manage to solve the issue? Looking at the video you shared, am facing exactly the same thing. The  simulation was working perfectly a week ago. It is even worse why the /robotnik_base_control/odom topic is empty.

grafoteka, 2019-06-21:
Hi @wendwosen, I'm sorry but I couldn't solve the problem

DeepRowLie, 2023-09-18:
I face the same problem with ROS neotic and ubuntu 20.04.In the simulation, after the robot has been turning in circles for a long time, issuing a new control command will result in a slight rotation.
I published Twist messages, where only the x-axis linear velocity is greater than zero. Below is the trajectory of the robot.
[image]
Below is my robot xacro
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="pioneer3dx_joints_state_publisher">
    <gazebo>
        <plugin name="joint_state_publisher" filename="libgazebo_ros_joint_state_publisher.so">
          <robotNamespace></robotNamespace>
          <jointName>chassis_swivel_joint, swivel_wheel_joint, left_hub_joint, right_hub_joint</jointName>
          <updateRate>50.0</updateRate>
          <alwaysOn>true</alwaysOn>
        </plugin>
    </gazebo>
  </xacro:macro>
  

  <xacro:macro name="pioneer3dx_diff_drive">
    <gazebo>
        <plugin name="differential_drive_controller" filename="libgazebo_ros_diff_drive.so">
            <rosDebugLevel>Debug</rosDebugLevel>
            <robotNamespace></robotNamespace>
            <publishWheelTF>false</publishWheelTF>
            <publishWheelJointState>true</publishWheelJointState>
            <alwaysOn>true</alwaysOn>
            <leftJoint>left_hub_joint</leftJoint>
            <rightJoint>right_hub_joint</rightJoint>
            <wheelSeparation>0.3</wheelSeparation>
            <wheelDiameter>0.18</wheelDiameter>
            <wheelTorque>20</wheelTorque>
            <wheelAcceleration>1.8</wheelAcceleration>
            <commandTopic>cmd_vel</commandTopic>
            <odometryTopic>odom</odometryTopic>
            <odometryFrame>odom</odometryFrame>
            <odometrySource>world</odometrySource>
            <robotBaseFrame>base_link</robotBaseFrame>
	    <updateRate>50</updateRate>
        </plugin>
    </gazebo>

  </xacro:macro>

</robot>