ros-simulation/gazebo_ros_pkgs #646: gazebo8 from ros-kinetic-gazebo8.. crashes when using compressed image transport

skohlbr, 2017-12-12:
I installed the full gazebo8 setup from OSRF's repo (i.e. `ros-kinetic-gazebo8*`) on a Kinetic/16.04 system. Everything works fine (love the responsive UI in 8), as long as one does not request a image using compressed image transport. If one does, gazebo instantly crashes. I managed to get some backtraces using Apport (from which one apparently cannot copy and paste for some weird reason [?]), so attached is a screenshot (4K ftw, need to zoom in a bit):
[image]

I guess this is some sort of OpenCV 2.4 vs 3.0 screw-up, but not sure about the details. What I tried was building
* vision_opencv
* image_transport_plugins
* gazebo_ros_pkgs

from source in my workspace, but this led to Gazebo instantly crashing on startup.

For a minimal example that can be used to reproduce the issue, a bare workspace with the package from https://github.com/skohlbr/gazebo_camera_timestamp_issue can be used. The launch file there fires up rviz with a ImageView and as soon as one selects "compressed" for transport instead of "raw", Gazebo crashes instantly.

@j-rivero  would be great if you could take a look :)

skohlbr, 2017-12-12:
https://github.com/ros-perception/image_common/issues/49 sounds fairly similar.

j-rivero, 2017-12-14:
> from source in my workspace, but this led to Gazebo instantly crashing on startup.

Without exploring the problem in runtime, the thing sound to me like an ABI problem somehow. However if you compiled `gazebo_ros_pkgs` and the error persists, that should not be the problem. Stefan do you get exactly the same error using packages and the catkin version or they are different?

For stopping apport to catch the traces I usually do:

```bash
sudo service apport stop
ulimit -c unlimited 
echo '/tmp/core.%e.%p.%t' | sudo tee /proc/sys/kernel/core_pattern
```
And look for the core file in /tmp. Does this work in your case?

skohlbr, 2017-12-16:
I added gazebo_ros_pkgs to my testing workspace (only containing the https://github.com/skohlbr/gazebo_camera_timestamp_issue repo otherwise), recompiled (using catkin_tools) and get the same error (i.e. gazebo crashes as soon as a compressed image is requested. Not sure what to make of this. Would be interesting to know if you (or anybody) can reproduce, would certainly expect that to be the case.
 
The core file instructions worked, but unlike with the apport report popping up in Ubuntu I don't get symbols shown when opening the file with gdb and doing a backtrace.

skohlbr, 2017-12-19:
FWIW, recently a lot of other issues popped up regarding Kinetic and OpenCV, https://github.com/ros-perception/vision_opencv/issues/193 appears to be a good "hub" on that.

ironmig, 2018-06-04:
I have replicated this. I have gazebo_ros_pkgs, image_transport_plugins, vision_opencv, and image_common all built from source (latest kinetic branch), and gazebo 8.4.0. Here is the stack trace
```
Thread 1 "gzserver" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/x86_64/strlen.S:106
106	../sysdeps/x86_64/strlen.S: No such file or directory.
(gdb) backtrace
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
#1  0x00007ffff5d41980 in printf_positional (s=s@entry=0x7ffffffec200, 
    format=format@entry=0x7fffd1910340 "%s:%d: error: (%d) %s in function %s\n", readonly_format=readonly_format@entry=0, ap=ap@entry=0x7ffffffec338, 
    ap_savep=ap_savep@entry=0x7ffffffebd98, done=41, nspecs_done=4, 
    lead_str_end=0x7fffd1910340 "%s:%d: error: (%d) %s in function %s\n", 
    work_buffer=0x7ffffffebdd0 "\005", save_errno=2, grouping=0x0, 
    thousands_sep=0x7ffff5e81b25 "") at vfprintf.c:2023
#2  0x00007ffff5d424a6 in _IO_vfprintf_internal (s=s@entry=0x7ffffffec200, 
    format=<optimized out>, 
    format@entry=0x7fffd1910340 "%s:%d: error: (%d) %s in function %s\n", 
    ap=ap@entry=0x7ffffffec338) at vfprintf.c:1677
#3  0x00007ffff5e0b754 in ___vsprintf_chk (
    s=0x7ffffffec350 ":-14960: error: (30) getMat_ in function ", flags=1, 
    slen=65536, 
    format=0x7fffd1910340 "%s:%d: error: (%d) %s in function %s\n", 
    args=0x7ffffffec338) at vsprintf_chk.c:82
#4  0x00007fffd17f81ec in cv::format[abi:cxx11](char const*, ...) ()
   from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
#5  0x00007fffd17f82ab in cv::Exception::formatMessage() ()
   from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
#6  0x00007ffef7315bc1 in cv::Exception::Exception(int, cv::String const&, cv::String const&, cv::String const&, int) ()
---Type <return> to continue, or q <return> to quit---
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#7  0x00007ffef731560c in cv::error(int, cv::String const&, char const*, char const*, int) () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#8  0x00007ffef736900a in cv::_InputArray::getMat_(int) const ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#9  0x00007ffefc110dbf in cv::imencode(cv::String const&, cv::_InputArray const&, std::vector<unsigned char, std::allocator<unsigned char> >&, std::vector<int, std::allocator<int> > const&) ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3
#10 0x00007ffef244afc2 in compressed_image_transport::CompressedPublisher::publish(sensor_msgs::Image_<std::allocator<void> > const&, boost::function<void (sensor_msgs::CompressedImage_<std::allocator<void> > const&)> const&) const (
    this=0x7ffef8015370, message=..., publish_fn=...)
    at /home/kallen/catkin_ws/src/image_transport_plugins/compressed_image_transport/src/compressed_publisher.cpp:125
#11 0x00007ffef2472afd in image_transport::SimplePublisherPlugin<sensor_msgs::CompressedImage_<std::allocator<void> > >::publish (this=0x7ffef8015370, 
    message=...)
    at /home/kallen/catkin_ws/src/image_common/image_transport/include/image_transport/simple_publisher_plugin.h:85
#12 0x00007ffefed6fdfc in image_transport::Publisher::publish (this=0x29d4be8, 
    message=...)
    at /home/kallen/catkin_ws/src/image_common/image_transport/src/publisher.cpp---Type <return> to continue, or q <return> to quit---
:165
#13 0x00007ffeffb52169 in gazebo::GazeboRosCameraUtils::PutCameraData (
    this=0x29d4ba8, 
    _src=0x31ca2c0 "\263"...)
    at /home/kallen/catkin_ws/src/gazebo_ros_pkgs/gazebo_plugins/src/gazebo_ros_camera_utils.cpp:631
```

You can see in the trace that some frames reference opencv 2 and some reference opencv 3, this is likely the problem. I'll look into how this happened... I don't think I did anything unusual with my install.

skohlbr, 2018-06-06:
Still also happens with current .debs. I reverted back to using Gazebo7 to avoid the issue.

ironmig, 2018-06-15:
Another update on this particular rabbit hole:

Installing gazebo8 with ```sudo apt install ros-kinetic-gazebo8``` triggers opencv2 (libopencv-core2.4v5) to be installed, because of gazebo 8's dependencies:
```
developer@3f754ae1222e:~/workspaces$ aptitude why libopencv-core2.4v5
i   ros-kinetic-gazebo8-ros Depends libgazebo8                       
i A libgazebo8              Depends libavdevice-ffmpeg56 (>= 7:2.4.3)
i A libavdevice-ffmpeg56    Depends libavfilter-ffmpeg5 (>= 7:2.4)   
i A libavfilter-ffmpeg5     Depends libopencv-core2.4v5    
```

Having both of these versions is likely causing the issue, likely because gazebo is linked with opencv2 while the plugin is linked with opencv3 (due to cv_bridge).

arixrobotics, 2018-06-25:
Hello all, 
I'm using Gazebo 9 (with Kinetic) and is also experiencing this Gazebo crash when requesting compressed image. Is there a fix yet?

ironmig, 2018-06-28:
There has not been a patch yet. One quick solution would be to run image_proc in the namespace of your gazebo camera, then subscribe to one of the compressed images that image_proc produces (image_color, image_rect_color, image_mono, image_rect_mono...)

skohlbr, 2018-11-13:
Just wanted to confirm that this also happens on Gazebo9 installed from OSRF .debs.