ros-simulation/gazebo_ros_pkgs #654: Segfault when URDF contains commented <plugin> tag

positron96, 2017-12-18:
I have an URDF file with <plugin> tag. If I comment it, a SEGFAULT happens when parsing a file.

ros kinetic (rosversion: 1.12.12)
Gazebo 7.0.0

Minimal files to reproduce the problem:
Launch file:

```xml
<?xml version="1.0" ?>
<launch>

	<include file="$(find gazebo_ros)/launch/empty_world.launch">
		<arg name="debug" value="true" />
	</include>

	<param name="robot_description" 
		textfile="ttt.urdf" />

	<node name="rover_spawn" pkg="gazebo_ros" type="spawn_model" 
		args="-urdf -param robot_description -model rover" >
	</node>
 
</launch>
```

URDF that is ok:

```xml
<?xml version="1.0"?>
<robot name="rover" xmlns:xacro="http://www.ros.org/wiki/xacro">

    <!--111plugin name="object_controller" filename="libgazebo_ros_planar_move.so">
    </plugin-->
 
</robot>
```

URDF that causes segfault:

```xml
<?xml version="1.0"?>
<robot name="rover" xmlns:xacro="http://www.ros.org/wiki/xacro">

    <!--plugin name="object_controller" filename="libgazebo_ros_planar_move.so">
    </plugin-->
 
</robot>
```

Note extra "111" in correct file.

Launching with problematic file with debug enabled gives this error:

```
Thread 13 "gzserver" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc4f19700 (LWP 1031)]
0x00007ffff3c069c6 in TiXmlNode::FirstChild(char const*) const ()
   from /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
(gdb) bt
#0  0x00007ffff3c069c6 in TiXmlNode::FirstChild(char const*) const ()
   from /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
#1  0x00007ffff3c06bce in TiXmlNode::FirstChildElement(char const*) const ()
   from /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
#2  0x00007fffcab7ddec in gazebo::GazeboRosApiPlugin::walkChildAddRobotNamespace(TiXmlNode*) () from //opt/ros/kinetic/lib/libgazebo_ros_api_plugin.so
#3  0x00007fffcab9dfe7 in gazebo::GazeboRosApiPlugin::spawnSDFModel(gazebo_msgs::SpawnModelRequest_<std::allocator<void> >&, gazebo_msgs::SpawnModelResponse_<std::allocator<void> >&) ()
   from //opt/ros/kinetic/lib/libgazebo_ros_api_plugin.so
#4  0x00007fffcab9eff5 in gazebo::GazeboRosApiPlugin::spawnURDFModel(gazebo_msgs::SpawnModelRequest_<std::allocator<void> >&, gazebo_msgs::SpawnModelResponse_<std::allocator<void> >&) ()
   from //opt/ros/kinetic/lib/libgazebo_ros_api_plugin.so
#5  0x00007fffcabd99fc in ros::ServiceCallbackHelperT<ros::ServiceSpec<gazebo_msgs::SpawnModelRequest_<std::allocator<void> >, gazebo_msgs::SpawnModelResponse_<std::allocator<void> > > >::call(ros::ServiceCallbackHelperCallParams&) ()
   from //opt/ros/kinetic/lib/libgazebo_ros_api_plugin.so
#6  0x00007fffca7e9bf1 in ros::ServiceCallback::call() ()
   from /opt/ros/kinetic/lib/libroscpp.so
#7  0x00007fffca83d210 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
#8  0x00007fffca83e683 in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/kinetic/lib/libroscpp.so
#9  0x00007fffcab7ca9a in gazebo::GazeboRosApiPlugin::gazeboQueueThread() ()
   from //opt/ros/kinetic/lib/libgazebo_ros_api_plugin.so
#10 0x00007ffff40d85d5 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#11 0x00007ffff65ed6ba in start_thread (arg=0x7fffc4f19700)
    at pthread_create.c:333
#12 0x00007ffff6bef3dd in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

j-rivero, 2017-12-18:
Thanks for the report. Do you think that could be the same problem described and fixed in https://github.com/ros-simulation/gazebo_ros_pkgs/pull/584?

positron96, 2017-12-19:
Yes, seems to be the same problem exactly! Does being fixed mean that I
could `apt-get upgrade` in near future to get a fix?

j-rivero, 2017-12-20:
> Does being fixed mean that I could `apt-get upgrade` in near future to get a fix?

Yes, in fact if you are using kinetic the fix was released in packages version 2.5.14. These packages are in the [shadow](http://wiki.ros.org/ShadowRepository) repository waiting for a Kinetic distribution sync. As a solution for now, you can use the shadow repository to get these packages or built the gazebo_ros_pkgs from source in a catkin workspaces since the kinetic-devel branch has the fix merged.

closed 2017-12-20 by j-rivero