ros-simulation/gazebo_ros_pkgs #666: gazebo_ros_control should allow multiple RobotHW plugins

mikepurvis, 2018-01-09:
Right now you can only specify one tag, however if would be nice to take advantage of the new [combined_robot_hw](http://wiki.ros.org/combined_robot_hw) to allow specifying multiple RobotHW plugins— my use case in particular would be a simulator where I want the behaviour of the DefaultRobotHW for joints/transmissions/actuators, but I also want another plugin that loads a different type of control interface altogether that has nothing to do with joints.

Right now I manage this by having my plugin extend DefaultRobotHW, but this shouldn't be necessary.

ironmig, 2018-05-29:
I was thinking of implementing this by creating ```CombinedRobotHWSim``` which will act like CombinedRobotHW but only allow loading of classes which extend RobotHWSim. Is this roughly what you meant?

bochen87, 2018-08-16:
This would be really good! Also, is it possible to then add a "blacklist" or "whitelist" of joints that should be powered / not powered by the DefaultRobotHW?

alextoind, 2021-02-17:
Any news on merging this feature?