ros-simulation/gazebo_ros_pkgs #667: Possible abi incompatibility: gazebo-ros-pkgs 2.5.14-1 / gazebo8-ros-pkgs 2.5.10-1

dhood, 2018-01-10:
I've come across a change in behaviour that was resolved by rebuilding from source. I suspect it was due to an abi incompatibility somewhere so am posting here for visibility.

Issue:
Contact with `gazebo_ros_control`'s controller manager from the `controller_manager` script in the `controller_manager` package causes it to hang. This is using ARIAC.

Expected behaviour:
Two controllers, `ariac/arm` and `joint_state_controller` are loaded and the spawners finish cleanly.
```
ROS_MASTER_URI=http://localhost:11311

process[relay_joint_states-1]: started with pid [30547]
process[startup_ariac-2]: started with pid [30559]
Ensuring scoring log file exists
File already exists: /root/.ariac/log/performance.log
[startup_ariac-2] process has finished cleanly
log file: /root/.ros/log/8c6f2d5a-f5a2-11e7-b4d6-0242ac110002/startup_ariac-2*.log
process[gazebo-3]: started with pid [30574]
process[spawn_gazebo_model-4]: started with pid [30579]
process[robot_state_publisher-5]: started with pid [30620]
process[fake_joint_calibration-6]: started with pid [30624]
process[joint_state_controller_spawner-7]: started with pid [30690]
process[arm_controller_spawner-8]: started with pid [30694]
SpawnModel script started
[INFO] [1515546417.887255, 0.000000]: Loading model XML from ros parameter
[INFO] [1515546417.890023, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
Warning [parser.cc:441] Converting a deprecated SDF source[data-string].
Warning [parser.cc:441] Converting a deprecated SDF source[data-string].
[INFO] [1515546418.796099, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1515546419.036203, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[INFO] [1515546419.036644, 0.000000]: Waiting for service /gazebo/set_model_configuration
[INFO] [1515546419.038172, 0.000000]: temporary hack to **fix** the -J joint position option (issue #93), sleeping for 1 second to avoid race condition.
Loaded ariac/arm
Loaded joint_state_controller
[INFO] [1515546429.047614, 0.000000]: Calling service /gazebo/set_model_configuration
[INFO] [1515546429.051047, 0.000000]: Set model configuration status: SetModelConfiguration: success
Started ['ariac/arm'] successfully
Started ['joint_state_controller'] successfully
[joint_state_controller_spawner-7] process has finished cleanly
log file: /root/.ros/log/8c6f2d5a-f5a2-11e7-b4d6-0242ac110002/joint_state_controller_spawner-7*.log
[arm_controller_spawner-8] process has finished cleanly
log file: /root/.ros/log/8c6f2d5a-f5a2-11e7-b4d6-0242ac110002/arm_controller_spawner-8*.log
[spawn_gazebo_model-4] process has finished cleanly
log file: /root/.ros/log/8c6f2d5a-f5a2-11e7-b4d6-0242ac110002/spawn_gazebo_model-4*.log
```

Actual behaviour:
After the load request of the first controller, the spawners hang in wait_for_service trying to contact the ControllerManager.

Affected:
- gazebo8 8.2.0-1 + ros-kinetic-gazebo8-ros-pkgs 2.5.10-1
- gazebo7 7.9.0-1 + ros-kinetic-gazebo-ros-pkgs 2.5.14-1

Solution:
Originally I thought it might have been similar https://github.com/ros-controls/ros_control/issues/265 but moving the two spawners into a single one didn't impact it (think the gazebo_ros_controller wouldn't suffer from that issue anyway).

Simply rebuilding `gazebo_ros_pkgs` from source fixed it in both cases.

j-rivero, 2018-01-10:
The general diagnostic makes a lot sense to me, specially if the fix is to rebuild the source from code. 

Only a little detail does not make sense to me: `-gazebo8-` wrappers are heavily affected by any chagen in the ABI of the dependencies, both in ROS and Gazebo itself, since it does not get built again when the ROS buildfarm change its dependencies. The problem would be the `ros-kinetic-gazebo-ros-pkgs` which should have been rebuilt on every change of its ROS dependencies. Given that the problem appears in both and the ABI checker in upstream Gazebo, I tend to think that the problem has been generated by a change in an ABI of the ROS dependencies but that should not affect `ros-kinetic-gazebo-ros-pkgs`.

I'm going to regenerate the `gazebo8`packages but we will need to find an explanation for the official gazebo7 wrappers.

dhood, 2018-01-10:
Since I was in "fix it" mode instead of "diagnose it" mode, I wasn't being particularly meticulous and it's _absolutely_ possible that something else is involved. Your reasoning about gazebo7 gazebo-ros-pkgs makes sense and makes me think something else was at work here, because I did experience the same behaviour with 7 and 8.

Now that I think about it, the container I was working in [had previously had gazebo-ros-pkgs installed from source](https://github.com/osrf/ariac-docker/pull/9/files#diff-44baf6f7c0d7d0beed7ccda1e6f7388bR83) into `/opt/ros/kinetic` which might have somehow interfered with the apt pkg installation update, and there might have been an abi break since that source installation some months ago. Perhaps that explains it, but even if it doesn't, I am happy to have gotten things working now, and no other users seem to have been affected, so this ticket is mostly for bookkeeping. If other users report the same thing it might be a current ABI issue, otherwise it seems more likely now that it's something unique to my situation, so I'll close this out 😄

closed 2018-01-10 by dhood