ros-simulation/gazebo_ros_pkgs #680: Timeout during some rostests while downloading models

scpeters, 2018-03-05:
Some display tests were recently enabled, but these are failing in some of the CI builds (for example, the [kinetic branch](https://build.osrfoundation.org/job/ros_gazebo_pkgs-ci-default_kinetic-xenial-amd64/95)). Looking in the console output, I see that many of these tests fail to complete before the 30 second timeout, but that they also have a console message saying:

~~~
[Wrn] [ModelDatabase.cc:339] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
~~~

Perhaps we should give a longer timeout or make sure that the models are downloaded ahead of time.

cc: @j-rivero

j-rivero, 2018-03-05:
The uncertain amount of time that Gazebo takes to start is a huge problem for several CI infrastructures including our own. We need to find a way to control it somehow or avoid long delays. A Gazebo command parameter, something like `--no-downloads`, that fails if models are not present in the system limiting all remote database download make some sense?

For the current scenario, another option instead of increasing test time would be to download and install gazebo_models into the testing system before running the tests. Steve, any preference between the two solutions?

Thanks!

scpeters, 2018-03-06:
I think I prefer the 2nd option (download models onto testing system), though we could consider adding a tool that would download all models that are `<include>`d from a given world file (such as the [checkerboard_plane from the distortion world](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/test/camera/distortion_barrel.world#L124)). Perhaps this kind of feature could be added to the `gz` tool?

j-rivero, 2018-03-06:
The option of the `--no-downloads` flags would be to guarantee that there is no connection from Gazebo to external server and fail if something is missing. This is probably not the wanted behavior for a normal user but really useful for CI.

Use `gz tool` to automatically pre-download all the necessary resources sounds good to me and it will create the perfect CI conditions in combination with the `--no-downloads`

scpeters, 2018-03-07:
I created a gazebo feature request for the `gz` tool to download models.

j-rivero, 2018-03-09:
We have merged the download and installation of gazebo_models as a tarball for the CI in this repo prior to run tests: https://bitbucket.org/osrf/release-tools/pull-requests/543