ros-simulation/gazebo_ros_pkgs #681: Test failure: gazebo_plugins.rosunit-distortion_barrel_test

scpeters, 2018-03-05:
The [barrel distortion test](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/test/camera/distortion_barrel.test) is [failing](https://build.osrfoundation.org/job/ros_gazebo8_pkgs-ci-pr_any_kinetic-xenial-amd64/70/testReport/junit/(root)/DistortionTest/cameraDistortionTest_2/). I see the following console output on my desktop machine (xenial, kinetic):

~~~
-- run_tests.py: execute commands
  /opt/ros/kinetic/share/rostest/cmake/../../../bin/rostest --pkgdir=/home/scpeters/ws/gazebo_ros_pkgs/src/gazebo_ros_pkgs/gazebo_plugins --package=gazebo_plugins --results-filename test_camera_distortion_barrel.xml --results-base-dir /data_fast/scpeters/ws/gazebo_ros_pkgs/build/gazebo_plugins/test_results /home/scpeters/ws/gazebo_ros_pkgs/src/gazebo_ros_pkgs/gazebo_plugins/test/camera/distortion_barrel.test 
... logging to /home/scpeters/.ros/log/rostest-t2-28237.log
[ROSUNIT] Outputting test results to /data_fast/scpeters/ws/gazebo_ros_pkgs/build/gazebo_plugins/test_results/gazebo_plugins/rostest-test_camera_distortion_barrel.xml
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1520029634.783625276]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1520029634.783996684]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.23.2.90
[ INFO] [1520029636.095044292]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1520029636.097818691]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1520029636.099974611]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ INFO] [1520029636.101283715]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[ INFO] [1520029636.205823815, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1520029636.241639053, 0.057000000]: Physics dynamic reconfigure ready.
[Testcase: testdistortion_barrel_test] ... ok

[ROSTEST]-----------------------------------------------------------------------

[gazebo_plugins.rosunit-distortion_barrel_test/cameraDistortionTest][FAILURE]---
gazebo_plugins/test/camera/distortion.cpp:167
Expected: (diff2) >= (diff1), actual: 174273819 vs 177511245
--------------------------------------------------------------------------------

[gazebo_plugins.rosunit-distortion_barrel_test/cameraDistortionTest][FAILURE]---
gazebo_plugins/test/camera/distortion.cpp:167
Expected: (diff2) >= (diff1), actual: 176783463 vs 177511245
--------------------------------------------------------------------------------

[gazebo_plugins.rosunit-distortion_barrel_test/cameraDistortionTest][FAILURE]---
gazebo_plugins/test/camera/distortion.cpp:173
Expected: (diff2) >= (diff1), actual: 177413820 vs 177511245
--------------------------------------------------------------------------------

[gazebo_plugins.rosunit-distortion_barrel_test/cameraDistortionTest][FAILURE]---
gazebo_plugins/test/camera/distortion.cpp:173
Expected: (diff2) >= (diff1), actual: 177420336 vs 177511245
--------------------------------------------------------------------------------

[gazebo_plugins.rosunit-distortion_barrel_test/cameraDistortionTest][FAILURE]---
gazebo_plugins/test/camera/distortion.cpp:167
Expected: (diff2) >= (diff1), actual: 177322950 vs 177511245
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1
~~~

These failed expectations indicate that the undistorted image doesn't seem to be the best match to the original image. I added the patch shown below to display the original and undistorted images, both for the pincushion and barrel distortion.

The fixed pincushion image (left side of screen) has some black bars, so it is cropped. The original and fixed images (right side of screen) match very closely.

[image]

The undistorted and fixed barrel images are on the bottom center and bottom right. The images don't have the black bars, but they also don't match each other well in size or shape.

[image]

~~~
diff --git a/gazebo_plugins/CMakeLists.txt b/gazebo_plugins/CMakeLists.txt
index 73e1986..6dc45ad 100644
--- a/gazebo_plugins/CMakeLists.txt
+++ b/gazebo_plugins/CMakeLists.txt
@@ -391,10 +391,10 @@ if (CATKIN_ENABLE_TESTING)
     add_rostest_gtest(distortion_barrel_test
                       test/camera/distortion_barrel.test
                       test/camera/distortion.cpp)
-    target_link_libraries(distortion_barrel_test ${catkin_LIBRARIES})
+    target_link_libraries(distortion_barrel_test ${catkin_LIBRARIES} opencv_highgui3)
     add_rostest_gtest(distortion_pincushion_test
                       test/camera/distortion_pincushion.test
                       test/camera/distortion.cpp)
-    target_link_libraries(distortion_pincushion_test ${catkin_LIBRARIES})
+    target_link_libraries(distortion_pincushion_test ${catkin_LIBRARIES} opencv_highgui3)
   endif()
 endif()
diff --git a/gazebo_plugins/test/camera/distortion.cpp b/gazebo_plugins/test/camera/distortion.cpp
index 30b0ad0..9e96650 100644
--- a/gazebo_plugins/test/camera/distortion.cpp
+++ b/gazebo_plugins/test/camera/distortion.cpp
@@ -1,6 +1,7 @@
 // OpenCV includes
 #include <opencv2/core/core.hpp>
 #include <opencv2/imgproc/imgproc.hpp>
+#include <opencv2/highgui/highgui.hpp>
 
 // Test includes
 #include <gtest/gtest.h>
@@ -133,6 +134,12 @@ TEST_F(DistortionTest, cameraDistortionTest)
   Mat fixed = distorted.clone();
   Mat undistorted = Mat(cv_bridge::toCvCopy(cam_image_undistorted_)->image);
 
+  cv::namedWindow("distorted", WINDOW_AUTOSIZE);
+  cv::imshow("distorted", distorted);
+
+  cv::namedWindow("undistorted", WINDOW_AUTOSIZE);
+  cv::imshow("undistorted", undistorted);
+
   //crop the image to remove black borders leftover from (un)distortion
   int cropBorder = 50;
   cv::Rect myROI(cropBorder, cropBorder,
@@ -142,6 +149,8 @@ TEST_F(DistortionTest, cameraDistortionTest)
 
 
   undistort(distorted, fixed, intrinsic_distorted_matrix, distortion_coeffs);
+  cv::namedWindow("fixed", WINDOW_AUTOSIZE);
+  cv::imshow("fixed", fixed);
 
   //Ensure that we didn't crop away everything
   ASSERT_GT(distorted.rows, 0);
@@ -156,6 +165,13 @@ TEST_F(DistortionTest, cameraDistortionTest)
   long diff1 = 0, diff2 = 0;
   diffBetween(fixed_crop, undistorted_crop, diff1);
 
+  cv::namedWindow("fixed_crop", WINDOW_AUTOSIZE);
+  cv::imshow("fixed_crop", fixed_crop);
+
+  cv::namedWindow("undistorted_crop", WINDOW_AUTOSIZE);
+  cv::imshow("undistorted_crop", undistorted_crop);
+  waitKey(0);
+
   const double OFFSET = 0.01;
 
   // test each parameter, varying one at a time
~~~

cc @Kukanani @iche033

Kukanani, 2018-03-07:
[The link you posted](https://build.osrfoundation.org/job/ros_gazebo8_pkgs-ci-pr_any_kinetic-xenial-amd64/70/testReport/junit/(root)/DistortionTest/cameraDistortionTest_2/) displays as "Passed" for me. However I can't see any build history, maybe a permissions issue?

But anyway, clearly from your screenshots the behavior is not correct. How long has the test been failing?

scpeters, 2018-03-07:
Weird, does this link show that a test failed?

* https://build.osrfoundation.org/job/ros_gazebo8_pkgs-ci-pr_any_kinetic-xenial-amd64/70/testReport/junit/(root)/DistortionTest/

I'm not sure how long it's been failing, since these tests weren't running CI. I don't know if it's a change in gazebo or in gazebo_ros_pkgs.

I have a workspace set up on my computer, so I can help with debugging this if you give me any suggestions. I guess I could try building an older version of gazebo to see if I can find a point when the test was passing?

iche033, 2018-03-07:
I don't think we made any changes to camera distortion in gazebo since the pincushion feature was added but I'll also investigate a little.

Kukanani, 2018-03-07:
So did the test ever run successfully in CI before now?

scpeters, 2018-03-07:
It was not run in CI before at all.

Also, I just compiled against gazebo 7.7, and the test is passing. So I think something changed in gazebo to make it fail. I'll try gazebo 7.8 next, then go until I can pinpoint what causes the failure.

scpeters, 2018-03-08:
It's weird. I can't reproduce the failure with a source build of gazebo7, even though I've seen a jenkins failure with the gazebo 7.11.0 deb:

* https://build.osrfoundation.org/job/ros_gazebo_pkgs-ci-pr_any_kinetic-xenial-amd64/87/testReport/junit/(root)/DistortionTest/

Looking through the console output, it's failing on the pincushion test, not the barrel.

scpeters, 2018-03-09:
I can reproduce the barrel failure with gazebo8 and gazebo9 debs

scpeters, 2018-03-09:
It's easier to see which tests are failing after giving them distinct names: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/685#issuecomment-371354835

Kukanani, 2018-03-21:
I haven't forgotten about this issue. I'll try to get my machine up and running for gazebo development soon and see if I can make any headway.

mjcarroll, 2018-04-11:
I'm starting to investigate this and I have additionally managed to reproduce the barrel failure with gazebo9 from source.

mjcarroll, 2018-05-02:
I have tracked down the issue here, it appears to be a change in a caching mechanism on the Gazebo end, I'll get a PR together shortly.

mjcarroll, 2018-05-30:
This ended up needing an upstream fix in Gazebo: 

https://bitbucket.org/osrf/gazebo/pull-requests/update/2963

mjcarroll, 2018-06-18:
@ironmig Didn't you fix this?

scpeters, 2020-04-29:
https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/2967/page/1 was merged and released in gazebo 9.2.0

scpeters, 2020-04-29:
we still need to add documentation for the `borderCrop` parameter