ros-simulation/gazebo_ros_pkgs #689: compile with gazebo9 error

jacknlliu, 2018-03-18:
I compile this package with gazebo9, and it prints following errors

```sh
-- +++ processing catkin package: 'gazebo_ros_control'
-- ==> add_subdirectory(gazebo_ros_pkgs/gazebo_ros_control)
-- Found DART: /usr/include (Required is at least version "6") found components:  dart 
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so  
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (Required is at least version "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.1.4
-- Checking for module 'uuid'
--   Found uuid, version 2.27.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 2.2.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 56.4.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavutil'
--   Found libavutil, version 54.31.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.2
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.6
-- Checking for module 'libzip'
--   Found libzip, version 1.0.1
-- Gazebo version: 9.0
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring done
CMake Error:
  Error evaluating generator expression:

    $<TARGET_PROPERTY:protobuf::libprotobuf,INTERFACE_INCLUDE_DIRECTORIES>

  Target "protobuf::libprotobuf" not found.

CMake Error:
  Error evaluating generator expression:

    $<TARGET_PROPERTY:protobuf::libprotobuf,INTERFACE_INCLUDE_DIRECTORIES>

  Target "protobuf::libprotobuf" not found.

CMake Error:
  Error evaluating generator expression:

    $<TARGET_PROPERTY:ignition-math4::ignition-math4,INTERFACE_INCLUDE_DIRECTORIES>

  Target "ignition-math4::ignition-math4" not found.

-- Generating done
-- Build files have been written to: /data/assembler/assembler-dynamic/build
Invoking "cmake" failed
```

jacknlliu, 2018-03-25:
Could anyone share something about how to compile this package with gazebo9 and dart dynamics engine?

jacknlliu, 2018-03-27:
@j-rivero Hi, I found you maintain the `ros-kinetic-gazebo9-ros-pkgs` from the package meta-information, did you meet this issue? Any suggestion for this? 
Thanks in advance!

jacknlliu, 2018-04-02:
I found the packages depend on gazebo_ros can't compile with this package, or it will get the previous error. But the `gazebo_ros_pkgs` can be compiled with gazebo and dart. 

I close this issue and I will have more exploration about this.

closed 2018-04-02 by jacknlliu

HappySamuel, 2019-04-25:
Hi @jacknlliu 

I faced the same problems as you. Have you found a solution for this? Can you share with me?

Best,
Samuel

j-rivero, 2019-04-25:
Never saw the problem before. Do you have installed protobuf and ignition-math4?

HappySamuel, 2019-04-26:
Hi @j-rivero 

Yes, i do have both protobuf and ignition-math4 installed. Looks like this is resulted by having Gazebo 9 with ROS Kinetic. For my case, although these errors pop out during catkin_make. But the catkin_make still can be done when you attempt to catkin_make again. Hope there's a way to solve this.

Best, 
Samuel

gpollayil, 2019-05-16:
@j-rivero 
Same error (as described by @HappySamuel ) happens for me on ROS Melodic with Gazebo 9.
Any updates on how to solve this?

jacknlliu, 2019-05-17:
@j-rivero Could you provide your procure to package it? Then we can follow your method and repeat it.

j-rivero, 2019-05-17:
>  Same error (as described by @HappySamuel ) happens for me on ROS Melodic with Gazebo 9.

could you please provide instructions about how to reproduce the problem? (platform used, versions used)

> @j-rivero Could you provide your procure to package it?

The [build.sh](https://build.osrfoundation.org/job/gazebo-ros-bloom-debbuilder/80/consoleFull#console-section-3) script in the buidlfarm could probably provide some hints about the packaging procedure for non official gazebo_ros_pkgs.

gpollayil, 2019-05-23:
> could you please provide instructions about how to reproduce the problem? (platform used, versions used)

I'm on ROS Melodic on Ubuntu 18.04 using Gazebo  9.0.0. You can reproduce the problem by cloning [Kuka LWR](https://github.com/CentroEPiaggio/kuka-lwr) into your catkin workspace, switching branch into `melodic-devel` and `catkin_make`. You will see that an error pops out during `catkin_make`, but the second `catkin_make` does not show the error anymore.

j-rivero, 2019-05-23:
Thanks @gpollayil for the instructions, I was able to reproduce it. Could you please try this patch:

```diff
diff --git a/lwr_controllers/CMakeLists.txt b/lwr_controllers/CMakeLists.txt
index 606db30..8f39100 100644
--- a/lwr_controllers/CMakeLists.txt
+++ b/lwr_controllers/CMakeLists.txt
@@ -20,6 +20,10 @@ find_package(catkin REQUIRED COMPONENTS
   tf_conversions
 )
 
+# bug in gazebo dependencies
+# See https://bitbucket.org/ignitionrobotics/ign-cmake/issues/40
+ign_import_target(UUID)
+
 add_definitions (-fpermissive -std=c++11)
 
 include_directories(include ${Boost_INCLUDE_DIR} ${catkin_INCLUDE_DIRS})
diff --git a/lwr_hw/CMakeLists.txt b/lwr_hw/CMakeLists.txt
index 325e6a0..1eb6f2b 100644
--- a/lwr_hw/CMakeLists.txt
+++ b/lwr_hw/CMakeLists.txt
@@ -15,6 +15,7 @@ find_package(catkin REQUIRED COMPONENTS
   pluginlib
   kdl_parser
   transmission_interface
+  gazebo_dev
   gazebo_ros
   std_msgs
 )
@@ -36,6 +37,7 @@ catkin_package(
     pluginlib
     kdl_parser
     transmission_interface
+    gazebo_dev
     gazebo_ros
     std_msgs
   INCLUDE_DIRS include
@@ -60,6 +62,10 @@ target_link_libraries(${PROJECT_NAME}
   ${catkin_LIBRARIES}
 )
 
+# bug in gazebo dependencies
+# See https://bitbucket.org/ignitionrobotics/ign-cmake/issues/40
+ign_import_target(UUID)
+
 ## REAL 
 # KUKA FRI library
 add_library(friremote
@@ -110,9 +116,6 @@ target_link_libraries(lwr_hw_fril_node
 
 ## SIMULATION
 # lwr hw sim gazebo plugin 
-find_package(gazebo REQUIRED)
-include_directories(${GAZEBO_INCLUDE_DIRS})
-link_directories(${GAZEBO_LIBRARY_DIRS})
 add_library(lwr_hw_gazebo_plugin  src/lwr_hw_gazebo_plugin.cpp
   include/${PROJECT_NAME}/lwr_hw_gazebo.hpp
 )
@@ -122,7 +125,6 @@ add_dependencies(lwr_hw_gazebo_plugin
 target_link_libraries(lwr_hw_gazebo_plugin
   ${PROJECT_NAME}
   ${catkin_LIBRARIES}
-  ${GAZEBO_LIBRARIES}
 )
 
 ## INSTALLATION
diff --git a/lwr_hw/package.xml b/lwr_hw/package.xml
index 0f418cf..55c314d 100644
--- a/lwr_hw/package.xml
+++ b/lwr_hw/package.xml
@@ -7,13 +7,13 @@
   <maintainer email="marco.esposito@tum.de">Marco Esposito</maintainer>
 
   <license>GPLv2</license>
-  
+
   <buildtool_depend>catkin</buildtool_depend>
 
   <depend>control_toolbox</depend>
   <depend>controller_interface</depend>
-  <depend>gazebo</depend>
   <depend>gazebo_ros</depend>
+  <depend>gazebo_dev</depend>
   <depend>hardware_interface</depend>
   <depend>transmission_interface</depend>
   <depend>realtime_tools</depend>
```

gpollayil, 2019-05-24:
Thank you @j-rivero ! The patch seems to do the job.

j-rivero, 2019-05-27:
> Thank you @j-rivero ! The patch seems to do the job.

would you mind sending it to upstream please?

gpollayil, 2019-08-30:
@j-rivero Do you mean, push the changes? 
P.S. Sorry for the delay: I completely missed your reply.  :/

j-rivero, 2019-09-05:
> @j-rivero Do you mean, push the changes?
> P.S. Sorry for the delay: I completely missed your reply. :/

No worries. I meant send my patch in the form of pull request to the upstream repository (in this case https://github.com/CentroEPiaggio/kuka-lwr).

gpollayil, 2019-09-09:
@j-rivero But I am not working on a fork... As I have access to the original repo and I have pushed the changes on the melodic-devel of the kuka-lwr repository... Is that ok for you?

j-rivero, 2019-09-09:
> @j-rivero But I am not working on a fork... As I have access to the original repo, I have pushed the changes on the melodic-devel of the kuka-lwr repository... Is that ok for you?

perfect, thanks!

marcingajewski14, 2019-11-30:
I had similar problem, added lines from patch and then there is issue like this. Can anyone help me with this one? @j-rivero  maybe You know what to do with this one? I will be really gratefeul for help. 

`marcin@marcin-VirtualBox:~/inz_ws$ catkin_make
Base path: /home/marcin/inz_ws
Source space: /home/marcin/inz_ws/src
Build space: /home/marcin/inz_ws/build
Devel space: /home/marcin/inz_ws/devel
Install space: /home/marcin/inz_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/marcin/inz_ws/build"
####
####
#### Running command: "make -j1 -l1" in "/home/marcin/inz_ws/build"
####
[  0%] Built target geometric_shapes
[  0%] Built target _moveit_msgs_generate_messages_check_deps_KinematicSolverInfo
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target shape_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _moveit_msgs_generate_messages_check_deps_CartesianTrajectory
[  0%] Built target _moveit_msgs_generate_messages_check_deps_JointLimits
[  0%] Built target _moveit_msgs_generate_messages_check_deps_ApplyPlanningScene
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PlanningSceneComponents
[  0%] Built target _moveit_msgs_generate_messages_check_deps_GetMotionPlan
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PickupGoal
[  0%] Built target _moveit_msgs_generate_messages_check_deps_GripperTranslation
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PlaceFeedback
[  0%] Built target _moveit_msgs_generate_messages_check_deps_SaveMap
[  0%] Built target _moveit_msgs_generate_messages_check_deps_GetPlanningScene
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PickupActionFeedback
[  0%] Built target _moveit_msgs_generate_messages_check_deps_ExecuteTrajectoryActionResult
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PlaceActionResult
[  0%] Built target _moveit_msgs_generate_messages_check_deps_ExecuteTrajectoryGoal
[  0%] Built target _moveit_msgs_generate_messages_check_deps_GetPositionFK
[  0%] Built target _moveit_msgs_generate_messages_check_deps_MotionPlanDetailedResponse
[  0%] Built target _moveit_msgs_generate_messages_check_deps_ListRobotStatesInWarehouse
[  0%] Built target _moveit_msgs_generate_messages_check_deps_MoveItErrorCodes
[  0%] Built target _moveit_msgs_generate_messages_check_deps_SaveRobotStateToWarehouse
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PlannerInterfaceDescription
[  0%] Built target _moveit_msgs_generate_messages_check_deps_Grasp
[  0%] Built target _moveit_msgs_generate_messages_check_deps_JointConstraint
[  0%] Built target _moveit_msgs_generate_messages_check_deps_LinkPadding
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PlaceLocation
[  0%] Built target _moveit_msgs_generate_messages_check_deps_RobotState
[  0%] Built target _moveit_msgs_generate_messages_check_deps_VisibilityConstraint
[  0%] Built target _moveit_msgs_generate_messages_check_deps_MoveGroupGoal
[  0%] Built target _moveit_msgs_generate_messages_check_deps_ExecuteTrajectoryAction
[  0%] Built target _moveit_msgs_generate_messages_check_deps_GetPositionIK
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PlaceResult
[  0%] Built target _moveit_msgs_generate_messages_check_deps_MoveGroupActionFeedback
[  0%] Built target _moveit_msgs_generate_messages_check_deps_OrientedBoundingBox
[  0%] Built target _moveit_msgs_generate_messages_check_deps_DisplayRobotState
[  0%] Built target _moveit_msgs_generate_messages_check_deps_MoveGroupAction
[  0%] Built target _moveit_msgs_generate_messages_check_deps_DeleteRobotStateFromWarehouse
[  0%] Built target _moveit_msgs_generate_messages_check_deps_CartesianPoint
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PositionIKRequest
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PickupActionResult
[  0%] Built target _moveit_msgs_generate_messages_check_deps_SetPlannerParams
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PickupActionGoal
[  0%] Built target _moveit_msgs_generate_messages_check_deps_QueryPlannerInterfaces
[  0%] Built target _moveit_msgs_generate_messages_check_deps_BoundingVolume
[  0%] Built target _moveit_msgs_generate_messages_check_deps_PositionConstraint
[  0%] Built target object_recognition_msgs_generate_messages_lisp
[  0%] Built target _moveit_msgs_generate_messages_check_deps_MoveGroupActionResult
[  0%] Built target _moveit_msgs_generate_messages_check_deps_ObjectColor
[  0%] Built target _moveit_msgs_generate_messages_check_deps_WorkspaceParameters
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target _moveit_msgs_generate_messages_check_deps_TrajectoryConstraints
[  0%] Built target _moveit_msgs_generate_messages_check_deps_GetRobotStateFromWarehouse
[  0%] Built target trajectory_msgs_generate_messages_lisp
[... 137 lines cut ...]
             ^~~~~~~~~~~~~~
/opt/ros/melodic/include/joint_limits_interface/joint_limits_urdf.h:48:13: note:   no known conversion for argument 1 from ‘const boost::shared_ptr<const urdf::Joint>’ to ‘urdf::JointConstSharedPtr {aka std::shared_ptr<const urdf::Joint>}’
/home/marcin/inz_ws/src/kuka-lwr/lwr_hw/src/lwr_hw.cpp:336:79: error: no matching function for call to ‘getSoftJointLimits(const boost::shared_ptr<const urdf::Joint>&, joint_limits_interface::SoftJointLimits&)’
[... cut ...] ptr<const urdf::Joint>}’
kuka-lwr/lwr_hw/CMakeFiles/lwr_hw.dir/build.make:62: recipe for target 'kuka-lwr/lwr_hw/CMakeFiles/lwr_hw.dir/src/lwr_hw.cpp.o' failed
make[2]: *** [kuka-lwr/lwr_hw/CMakeFiles/lwr_hw.dir/src/lwr_hw.cpp.o] Error 1
CMakeFiles/Makefile2:6990: recipe for target 'kuka-lwr/lwr_hw/CMakeFiles/lwr_hw.dir/all' failed
make[1]: *** [kuka-lwr/lwr_hw/CMakeFiles/lwr_hw.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
`

KerolousMomtaz, 2020-07-29:
@j-rivero
I'm on ROS Melodic on Ubuntu 18.04 using Gazebo 9.0.0 . i'm trying to run catkin_make and i get that 

-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.7
-- Checking for module 'libzip'
--   Found libzip, version 1.1.2
-- +++ processing catkin package: 'pal_gripper_description'
-- ==> add_subdirectory(tiago_description_complete/pal_gripper/pal_gripper_description)
-- +++ processing catkin package: 'pal_wsg_gripper_description'
-- ==> add_subdirectory(tiago_description_complete/pal_wsg_gripper/pal_wsg_gripper_description)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kora/simulator/sim_env_ws/build
####
#### Running command: "make -j1 -l1" in "/home/kora/simulator/sim_env_ws/build"
####
[  0%] Built target gazebo_msgs_generate_messages_py
[  0%] Built target trajectory_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target trajectory_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target tf_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_py
[  0%] Built target gazebo_msgs_generate_messages_eus
[  0%] Built target tf_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target gazebo_ros_gencfg
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_srvs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_lisp
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_nodejs
[  0%] Built target gazebo_msgs_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target gazebo_msgs_generate_messages_lisp
[  0%] Built target gazebo_msgs_generate_messages_nodejs
[ 33%] Linking CXX shared library /home/kora/simulator/sim_env_ws/devel/lib/librealsense_gazebo_plugin.so
[100%] Built target realsense_gazebo_plugin
[100%] Built target pal_gripper_description_xacro_generated_to_devel_space_
[100%] Built target pal_wsg_gripper_description_xacro_generated_to_devel_space_