ros-simulation/gazebo_ros_pkgs #696: Release for Melodic

j-rivero, 2018-04-04:
Hi all:

It is time to think on creating the `melodic-devel` branch from `lunar-devel`. Disruptive changes can be made at this time like those that modifies the API and the topic/service interface. Please speak up if you have in mind anything that needs to be merged.

We will need `ros_controls` but I believe that @bmagyar will be taking care of it during the next days.

bmagyar, 2018-04-06:
We've been ready for a while now ;)

j-rivero, 2018-04-09:
Currently block by the lack of some dependencies: `ros-melodic-cv-bridge, ros-melodic-diagnostic-updater, ros-melodic-pcl-conversions`:
 * cv-bridge: https://github.com/ros-perception/vision_opencv/issues/208
 * pcl-conversions: https://github.com/ros-perception/pcl_conversions/issues/28
 * diagnostic-update: https://github.com/ros/diagnostics/issues/81