ros-simulation/gazebo_ros_pkgs #717: gazebo_ros_diff_drive in encoder mode sends velocity in odom not base_link

KrzysztofZurad, 2018-05-16:
I realized that diff_drive_plugin in **encoder mode** sends velocity in the odom frame not base_link frame. I am a little bit confused. I've been trying to fuse data with robot_localization package using only velocity information from odometry and imu, but It won't work since the velocities are reported in wrong frame.

With parameter odometry_source set to WORLD we get:

```
    // convert velocity to child_frame_id (aka base_footprint)
    float yaw = pose.Rot().Yaw();
    odom_.twist.twist.linear.x = cosf ( yaw ) * linear.X() + sinf ( yaw ) * linear.Y();
    odom_.twist.twist.linear.y = cosf ( yaw ) * linear.Y() - sinf ( yaw ) * linear.X();`
```

but with odometry_source set to ENCODER we get:

```
    double dx = ( ssum ) /2.0 * cos ( pose_encoder_.theta + ( sdiff ) / ( 2.0*b ) );
    double dy = ( ssum ) /2.0 * sin ( pose_encoder_.theta + ( sdiff ) / ( 2.0*b ) );
    double dtheta = ( sdiff ) /b;

    double w = dtheta/seconds_since_last_update;
    double v = sqrt ( dx*dx+dy*dy ) /seconds_since_last_update;

    odom_.twist.twist.angular.z = w;
    odom_.twist.twist.linear.x = dx/seconds_since_last_update;
    odom_.twist.twist.linear.y = dy/seconds_since_last_update;
```

I would suggest to change it to 

```
    odom_.twist.twist.angular.z = w;
    odom_.twist.twist.linear.x = v;
    odom_.twist.twist.linear.y = 0;
```

ironmig, 2018-05-24:
I agree. This is inconsistent and against the message definition of nav_msgs/Odometry which states that twist will be in child frame Id. I will PR this change, thanks for spotting.

j-rivero, 2018-05-25:
Fixed in melodic-devel by #719. For other branches, please see [my comment](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/719#issuecomment-392089130) about the legacy protection.

closed 2018-05-25 by j-rivero