ros-simulation/gazebo_ros_pkgs #725: Disruptive changes in Melodic

j-rivero, 2018-05-25:
Here is the list of disruptive changes to be commited into Melodic that could require user reaction. Sorry for not doing them before the final release but I think it is a good time to made them:

### Joint Pose Trajectory
 - [ ] Remove the `gazebo_ros_joint_pose_trajectory` item (pr #708)
### Gazebo ROS Control
 - [ ] Remove legacy option from `gazebo_ros_control` in order not to ignore robotNamespace (pr #637, pr #709) 
### Diff Drive plugin
 - [x] Change encoder mode to send velocity from odom frame to base_link frame (#issue #717, pr #719)
 - [ ] Remove legacy protection about the inversion of right and left wheels in code (pr #707)

ironmig, 2018-05-25:
Recommend adding 
[ ] remove ```-gazebo``` option from spawnmodel (deprecated in ROS Hydro), issue #729

davetcoleman, 2018-06-14:
Thanks @j-rivero for taking care of these migration changes, very important!

ironmig, 2018-07-06:
All items are now merged in melodic.

closed 2018-07-06 by ironmig