ros-simulation/gazebo_ros_pkgs #752: gazebo_ros_control_plugin - incorrect information on URDF parameter

msadowski, 2018-06-25:
Hi!

I'm on ROS kinetic (all packages installed through apt-get). And I run into a problem where I wouldn't be able to start diff_drive_controller as Gazebo was showing me an error:

```
[ERROR] [1529904002.814702511, 5.676000000]: Robot descripion couldn't be retrieved from param server.
[ERROR] [1529904002.814760134, 5.676000000]: Failed to initialize the controller
[ERROR] [1529904002.814814065, 5.676000000]: Initializing controller 'mobile_base_controller' failed
```

Even though gazebo_ros_control displays this message when it's starting:

`[ INFO] [1529903997.363384038, 0.360000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.`

I believe it is not correct because once I remapped my namespaces so that my robot description was in parameter with /my_namespace/robot_description things started working correctly. In short, having my robot description in /robot_description wasn't working, while having it in /some_namespace/robot_description is working perfectly.

If you would like to quickly check it I made a [repo](https://github.com/msadowski/dd_bot) where I was experimenting (47b8c9d662 is the last commit, where the issue can be seen, I think).

ironmig, 2018-06-26:
Looks like you specify ```<robotNamespace>``` in your gazebo_ros_control plugin definition. This is the namespace robot_description is expected to be in.  You can also override this by setting ```<robotParam>``` to the parameter you set your robot description to.

msadowski, 2018-06-27:
@ironmig That's what I thought. I'm wondering whether it's possible to include the <robotNamespace> parameter in the message visible from Gazebo? I think this would make it much clearer