ros-simulation/gazebo_ros_pkgs #758: Fish eye camera plugin for ROS

sychaichangkun, 2018-07-01:
Hi, I want to simulate fish eye camera in Gazebo and publish image via ROS.
I find a tutorial to [create fish eye cam in Gazebo](http://gazebosim.org/tutorials?tut=wide_angle_camera&branch=wideanglecamera) , but  there seems no ROS plugin.

How can I write a plugin which converts Gazebo messages to ROS? 
I need your help. Thanks.

ironmig, 2018-07-02:
Have you tried adding the gazebo_ros_camera plugin to your sdf like [in this tutorial](http://gazebosim.org/tutorials?tut=ros_gzplugins#Camera)?

sychaichangkun, 2018-07-05:
Yes, I tried but gazebo_ros_camera doesn't support fish eye, when I changed the FOV to more than 180 degrees, it failed.

ironmig, 2018-07-05:
What do you mean by "it failed". Just ran gazebo with this sensor config like in the tutorial and it worked as expected:
```
<sensor name="camera" type="wideanglecamera">
  <camera>
    <horizontal_fov>6.283</horizontal_fov>
    <image>
      <width>320</width>
      <height>240</height>
    </image>
    <clip>
      <near>0.1</near>
      <far>100</far>
    </clip>
    <!-- A new section named `lens`. -->
    <lens>
      <!-- type element is mandatory -->
      <type>custom</type>
      <!-- manually defined mapping function r = c1*f*fun(theta/c2 + c3) -->
      <!-- More information here: https://en.wikipedia.org/wiki/Fisheye_lens#Mapping_function -->
      <custom_function>
        <c1>1.05</c1>
        <!-- linear scaling -->
        <c2>4</c2>
        <!-- angle scaling -->
        <f>1.0</f>
        <!-- one more scaling parameter -->
        <fun>tan</fun>
        <!-- one of sin,tan,id -->
      </custom_function>
      <!-- if it is set to `true` your horizontal FOV will ramain as defined -->
      <!-- othervise it depends on lens type and custom function, if there is one -->
      <scale_to_hfov>true</scale_to_hfov>
      <!-- clip everything that is outside of this angle -->
      <cutoff_angle>3.1415</cutoff_angle>
      <!-- resolution of the cubemap texture, the highter it is - the sharper is your image -->
      <env_texture_size>512</env_texture_size>
    </lens>
    <always_on>1</always_on>
    <update_rate>30</update_rate>
  </camera>
  <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
    <cameraName>camera1</cameraName>
    <imageTopicName>image_raw</imageTopicName>
    <cameraInfoTopicName>camera_info</cameraInfoTopicName>
    <frameName>camera_link</frameName>
    <hackBaseline>0.07</hackBaseline>
  </plugin>
</sensor>
```
[image]

carolina68, 2019-05-12:
Hi! I am now also working on the simulation of fish-eye cameras in Gazebo. After follwing the tutorial mentioned by you(no modification by myself), there alwys exists an error in simulation when I increse the FOV to lager than 180 degrees:
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.

Do you have any idea what the problem may be?
Thanks in advance.

chapulina, 2019-05-12:
@carolina68 , that could be related to this open issue: https://bitbucket.org/osrf/gazebo/issues/2558/

chakloong883, 2025-02-28:
> What do you mean by "it failed"? Just ran gazebo with this sensor config like in the tutorial and it worked as expected:
> 
> ```
> <sensor name="camera" type="wideanglecamera">
>   <camera>
>     <horizontal_fov>6.283</horizontal_fov>
>     <image>
>       <width>320</width>
>       <height>240</height>
>     </image>
>     <clip>
>       <near>0.1</near>
>       <far>100</far>
>     </clip>
>     
>     <lens>
>       
>       <type>custom</type>
>       
>       
>       <custom_function>
>         <c1>1.05</c1>
>         
>         <c2>4</c2>
>         
>         <f>1.0</f>
>         
>         <fun>tan</fun>
>         
>       </custom_function>
>       
>       
>       <scale_to_hfov>true</scale_to_hfov>
>       
>       <cutoff_angle>3.1415</cutoff_angle>
>       
>       <env_texture_size>512</env_texture_size>
>     </lens>
>     <always_on>1</always_on>
>     <update_rate>30</update_rate>
>   </camera>
>   <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
>     <cameraName>camera1</cameraName>
>     <imageTopicName>image_raw</imageTopicName>
>     <cameraInfoTopicName>camera_info</cameraInfoTopicName>
>     <frameName>camera_link</frameName>
>     <hackBaseline>0.07</hackBaseline>
>   </plugin>
> </sensor>
> ```
> 
> [image]

Hi，I have tried your implement, it worked! But the imaging circle is tangent to the height of the image, I want to modify it to be tangent to the width. How should I adjust the parameters?