ros-simulation/gazebo_ros_pkgs #768: gazebo_ros_laser and gazebo_ros_gpu_laser incompatibility

AntoineRichard, 2018-07-12:
I try to use both gpu_ray and ray sensors at the same time but it always gets me an error which leads to the none of the laser_scans being published. I am using Gazebo 9.1 and ros-kinetic (I have a Nvidia 1060GTX GPU). The sensor are working individually, and I can add mulitple cpu or gpu sensors as long as I only use one type of sensors in the scene.

To test the issue I first launch an empty gazebo world using:
` roslaunch gazebo_ros empty_world.launch`
Then I spawn the model a modified turtlebot from turtlebot_gazebo and I get the following error:

In the gazebo_ros launch:
```
NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)<
auto-starting new master
process[master]: started with pid [22009]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d4f820bc-85c2-11e8-a8d8-9cb6d01b13b7
process[rosout-1]: started with pid [22033]
started core service [/rosout]
process[gazebo-2]: started with pid [22042]
process[gazebo_gui-3]: started with pid [22062]
[ INFO] [1531393248.650194823]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1531393248.650645568]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1531393249.665060561, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1531393249.692855553, 0.049000000]: Physics dynamic reconfigure ready.
[ INFO] [1531393256.792236413, 6.752000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1531393256.796050738, 6.752000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1531393256.859807230, 6.752000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1531393256.859898936, 6.752000000]: Starting GazeboRosLaser Plugin (ns = /)
[ INFO] [1531393256.861411545, 6.752000000]: GPU Laser Plugin (ns = /) <tf_prefix_>, set to ""
[ INFO] [1531393256.862300586, 6.752000000]: LoadThread function completed
Segmentation fault (core dumped)
[gazebo-2] process has died [pid 22042, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode worlds/empty.world __name:=gazebo __log:=/home/antoine/.ros/log/d4f820bc-85c2-11e8-a8d8-9cb6d01b13b7/gazebo-2.log].
log file: /home/antoine/.ros/log/d4f820bc-85c2-11e8-a8d8-9cb6d01b13b7/gazebo-2*.log
 ```
Unfortunately I cannot find the log file....

In the spawn model launch:
```
process[spawn_turtlebot_model-1]: started with pid [22436]
process[mobile_base_nodelet_manager-2]: started with pid [22437]
process[cmd_vel_mux-3]: started with pid [22438]
process[bumper2pointcloud-4]: started with pid [22440]
[ERROR] [1531393257.278663, 6.752000]: Unpause physics service call failed: unable to connect to service: [Errno 104] Connection reset by peer
[spawn_turtlebot_model-1] process has finished cleanly
log file: /home/antoine/.ros/log/d4f820bc-85c2-11e8-a8d8-9cb6d01b13b7/spawn_turtlebot_model-1*.log
```
Logfile:
```
[rosout][INFO] 2018-07-12 13:00:57,127: Spawn status: SpawnModel: Successfully spawned entity
[rospy.internal][WARNING] 2018-07-12 13:00:57,278: Unknown error initiating TCP/IP socket to antoine-Blade:44263 (http://antoine-Blade:38983/): Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/network.py", line 362, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

[rospy.internal][WARNING] 2018-07-12 13:00:57,278: Unknown error initiating TCP/IP socket to antoine-Blade:44263 (rosrpc://antoine-Blade:44263): Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/network.py", line 362, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

[rospy.internal][INFO] 2018-07-12 13:00:57,278: topic[/clock] removing connection to http://antoine-Blade:38983/
[rosout][ERROR] 2018-07-12 13:00:57,278: Unpause physics service call failed: unable to connect to service: [Errno 104] Connection reset by peer
[rospy.core][INFO] 2018-07-12 13:00:57,279: signal_shutdown [atexit]
[rospy.internal][INFO] 2018-07-12 13:00:57,281: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2018-07-12 13:00:57,281: atexit
```

I need this solution because I have to label human being laser hits and the GPU sensors doesn't support laser retro value while the CPU sensor does not sense the "skin" of the gazebo actors. With the 2 sensors a simple point to point subtraction allows to find the laser rays that touches humans.

ironmig, 2018-07-13:
Have you tried running this setup without the gazebo_ros plugins (just the gazebo sensors)? I want to identify if this is an issue in Gazebo or in the ros plugins in this repository.

AntoineRichard, 2018-07-18:
I downloaded the sources and replaced GazeboRosLaser by GazeboRosGPULaser in "gazebo_plugins/src/gazebo_ros_gpu_laser.cpp" and "gazebo_plugins/include/gazebo_plugins/gazebo_ros_gpu_laser.h", and it's working fine. The classes had the same name in "gazebo_ros_gpu_laser" and gazebo_ros_laser" which was responsible for the issue.

ironmig, 2018-07-18:
Great job finding that! Please consider submitting a pull request with you fix

Darkproduct, 2022-01-07:
This seems like that it can be closed.