ros-simulation/gazebo_ros_pkgs #772: Combine gazebo_ros_laser, gazebo_ros_block_laser, and gazebo_ros_gpu_laser

ironmig, 2018-07-19:
Similar to the DepthCamera/OpenniKinect, these plugins are nearly identical which makes maintenance difficult and confuses users. 

I propose we combine the three into one plugin ```gazebo_ros_ray_sensor``` (following the naming in gazebo), which will provide feature parity to the three plugins through sdf configuration. 

A couple ideas for the design
* remove the Gaussian Noise feature,  this feature is already in gazebo and should not be duplicated in the gazebo_ros plugin
* remove the update_rate feature, this feature is already in gazebo and should not be duplicated in the gazebo_ros plugin
* remove the code related to tf_prefix, this feature was deprecated in tf long ago
* Will work as a plugin for either a RaySensor or GpuRaySensor. It can simply check which type it is on Load
* Allow the user to customize what kind(s) of ros message to output for the sensor, either LaserScan, PointCloud, or PointCloud2. 
* include a min_intensity option, which is currently only available in block_laser

Referencing the following issues which will be affected/resolved by this:
#50 
#133 
#333 
#768

ironmig, 2018-07-20:
There's another one... gazebo_ros_range which can be merged into this new one.

arixrobotics, 2018-08-13:
Hi, 
Is this branch `ray-plugin` https://github.com/ros-simulation/gazebo_ros_pkgs/tree/ray-plugin working? 
I'm looking for a 'gpu-version' of the `gazebo_ros_block_laser` plugin so that it can detect Gazebo 8/9 actors. If I can use your new `gazebo_ros_ray_sensor`, that'll be awesome. Any pointers on how to set it up?

ironmig, 2018-08-16:
Yes it should work fine, I've used it in my machine but have not written an automated test yet. You can just clone that branch into your catkin workspace. Your SDF will look something like this:

```
<link name="block_laser_link">
  <!-- Visuals / Collisions omitted for this example -->
  <sensor type="gou_ray" name="block_laser_sensor">
    <ray>
      <scan>
        <horizontal>
          <samples>10</samples>
          <resolution>1</resolution>
          <min_angle>-0.523</min_angle>
          <max_angle>0.523</max_angle>
        </horizontal>
        <vertical>
          <samples>10</samples>
          <resolution>1</resolution>
          <min_angle>-0.523</min_angle>
          <max_angle>0.523</max_angle>
        </vertical>
      </scan>
      <range>
        <min>0.10</min>
        <max>5.0</max>
        <resolution>0.01</resolution>
      </range>
    </ray>
    <plugin name="my_ray_sensor_plugin" filename="libgazebo_ros_ray_sensor.so">
      <topicName>/rrbot/laser/pointcloud</topicName>
      <frameName>block_laser_link</frameName>
      <!-- Can also use sensor_msgs/PointCloud if you prefer -->
      <outputType>sensor_msgs/PointCloud2</outputType>
    </plugin>
  </sensor>
</link>
```

arixrobotics, 2018-08-17:
Yes the plugin worked. Thank you. 
Unfortunately, Gazebo 9 doesn't support multiple GPU-type vertical rays yet :(

igricart, 2021-03-30:
Hi there!
Based on [that](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/778) is this issue also supposed to be closed?

@kev-the-dev Thanks very much for providing this plugin!