ros-simulation/gazebo_ros_pkgs #786: crash: libgazebo_ros_api_plugin.dylib

horagong, 2018-08-01:
I installed gazebo 8.6.0 on mac.

It crashes when doing roslaunch gazebo_ros empty_world.launch.

```
gzserver -s /opt/ros/lunar/lib/libgazebo_ros_api_plugin.dylib --verbose
Gazebo multi-robot simulator, version 8.6.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1533113914.954476000]: Finished loading Gazebo ROS API Plugin.
[Msg] Waiting for master.
[ INFO] [1533113914.955681000]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.26.121.251
[ INFO] [1533113915.989310000, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1533113916.058280000, 0.088000000]: Physics dynamic reconfigure ready.
Segmentation fault: 11
```

ironmig, 2018-08-01:
I don't have access to a mac to reproduce this... could you try to get a backtrace?

```roslaunch gazebo_ros empty_world.launch debug:=true```

horagong, 2018-08-02:
I upgraded ogre1.9 to 1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_6 and it's ok now.

closed 2018-08-06 by ironmig