ros-simulation/gazebo_ros_pkgs #788: Video Plugin Crashes Gzserver (Gazebo9)

arixrobotics, 2018-08-02:
Hello all,

It seems that the video plugin `libgazebo_ros_video.so` is crashing gzserver everytime I load it in ROS Kinetic + Gazebo 9.

This is the code snippet:
 ``` 
<gazebo reference="screen">
<visual>
  <plugin name="screen_video_controller" filename="libgazebo_ros_video.so">
    <height>600</height>
    <width>1024</width>
    <topicName>/robot/head_display</topicName>
    <robotNamespace>/robot</robotNamespace>
  </plugin>
</visual>
</gazebo>
```
If I try to load this model, I'll get:
```
...
[ INFO] [1533119655.853751756]: GazeboRosVideo (gzclient, ns = /robot) has started
Segmentation fault (core dumped)
[gazebo_gui-3] process has died [pid 12932, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/arif/.ros/log/6157ef48-9576-11e8-90e1-6c71d97cbaff/gazebo_gui-3.log].
log file: /home/arif/.ros/log/6157ef48-9576-11e8-90e1-6c71d97cbaff/gazebo_gui-3*.log
```
But if I comment out that part, it loads ok (without that video display, of course).

I've seen this issue, but it seems like a different case https://github.com/ros-simulation/gazebo_ros_pkgs/issues/354 

Can anyone help?

ironmig, 2018-08-02:
Thanks for the report. I'm curious if this is related to #646. Are you able to test this with gazebo7 + Kinetic?

arixrobotics, 2018-08-02:
I've tried on Kinetic + Gazebo 7, it works fine.

I'll try to check on the cv2/3 clash and see if that's the issue.

shiveshkhaitan, 2019-03-11:
Building ```gazebo_ros_video``` and ```cv_bridge``` from source works fine.