ros-simulation/gazebo_ros_pkgs #797: [ros2] Plugins not creating their own node

dhood, 2018-08-10:
There is overhead associated with having multiple nodes, e.g. services for parameters. This is especially true at the moment because ROS 2 nodes each have their own DDS participant.

Currently plugins are creating their own nodes, which is useful for isolating their remapping arguments etc. Eventually we would like to use a similar concept to a NodeHandle in ROS 1, wherein each plugin can have access to a namespaced node-like interface that does not require the creation of a new node underneath. In the meantime, we can keep using one node per plugin in a way that anticipates this change: plugins will be able to assume that they have access to an environment that is uniquely its own, e.g. no need to create subloggers or namespace parameters.

chapulina, 2018-08-15:
Action items from last week's discussion with @dhood , @ironmig and @wjwwood:

- [ ] Change `GazeboRosNode::Create()` to `GazeboRosNode::Get()`, because in the future the function may return a pre-existing node instead of creating a new one
- [ ] There will be two `Get`s, one which takes an SDF pointer, and another which returns a global static node
- [ ] Remove node name parameter when creating the node, instead use the plugin name from SDF (`<plugin name="this_one" ...>`)
- [ ] Remove `<node_name>` from SDF

Other conclusions for now:

* Keep returning one node per plugin until we hit a performance issue.
* Keep `<argument ...` tags for now, but later we may handle remapping differently
* In the future, `GazeboRosNode` may inherit from something like an `rclcpp::NodeHandle`

wjwwood, 2018-08-15:
> In the future, `GazeboRosNode` may inherit from something like an `rclcpp::NodeHandle`

Just to check my understanding, it currently inherits from `rclcpp::Node`?

chapulina, 2018-08-15:
> Just to check my understanding, it currently inherits from rclcpp::Node?

Correct, [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_ros/include/gazebo_ros/node.hpp#L36)'s the implementation. @tfoote also mentioned that in the future `rclcpp::Node` itself may behave like `ros::NodeHandle`, in which case `gazebo_ros_pkgs` may get away without changing anything.

eborghi10, 2019-03-04:
[The last item of this list](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-integration#open_file_folder-gazebo_ros) should be marked as done, right?

chapulina, 2019-03-04:
Nice catch, @eborghi10 . I [updated](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-integration/_compare/0d29384a890a874d86c66732df329438599571d8...20df101bc1b49957a0493d236b0ecac8e3bc0015) that document.