ros-simulation/gazebo_ros_pkgs #799: [ros2] Sensor noise not loaded in tests using ServerFixture

dhood, 2018-08-17:
This world file specifies noise for an IMU sensor:

[gazebo_ros_imu_sensor.world.txt](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2298904/gazebo_ros_imu_sensor.world.txt)

If it's loaded via the commandline with
```
gazebo ~/ros2_ws/src/ros2/gazebo_ros_pkgs/gazebo_plugins/test/worlds/gazebo_ros_imu_sensor.world --verbose
```
the following is output to ~/.gazebo/server-11345/default.log (the sensor noise is applied):
```
(1534544514 841628113) [Msg] Waiting for master.
(1534544514 849987949) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1534544514 850345075) [Msg] Publicized address: 172.23.1.168
(1534544515 486627737) applying Gaussian noise model with mean 0, stddev 0.0002, bias -9.08301e-06
(1534544515 486710876) applying Gaussian noise model with mean 0, stddev 0.0002, bias 7.48293e-06
(1534544515 486758102) applying Gaussian noise model with mean 0, stddev 0.0002, bias 7.44982e-06
(1534544515 486800228) Applying angular velocity noise to IMU[my_imu].
  X: Gaussian noise, mean[0], stdDev[0.0002] bias[-9.08301e-06] precision[0] quantized[0]
  Y: Gaussian noise, mean[0], stdDev[0.0002] bias[7.48293e-06] precision[0] quantized[0]
  Z: Gaussian noise, mean[0], stdDev[0.0002] bias[7.44982e-06] precision[0] quantized[0]
(1534544515 486816156) applying Gaussian noise model with mean 0, stddev 0.017, bias -0.100989
(1534544515 486851102) applying Gaussian noise model with mean 0, stddev 0.017, bias 0.100486
(1534544515 486875241) applying Gaussian noise model with mean 0, stddev 0.017, bias -0.0996727
(1534544515 486897781) Applying linear acceleration noise to IMU[my_imu].
  X: Gaussian noise, mean[0], stdDev[0.017] bias[-0.100989] precision[0] quantized[0]
  Y: Gaussian noise, mean[0], stdDev[0.017] bias[0.100486] precision[0] quantized[0]
  Z: Gaussian noise, mean[0], stdDev[0.017] bias[-0.0996727] precision[0] quantized[0]
(1534544515 487859675) Init world[gazebo_ros_imu_sensor_world]
```

However, if the world file is loaded from a test using ServerFixture with `this->Load("worlds/gazebo_ros_imu_sensor.world", true);`, the log has the following:

```
(1534543567 401762859) [Msg] Waiting for master.
(1534543567 410273917) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1534543567 410683769) [Msg] Publicized address: 172.23.1.168
(1534543568 98818777) Init world[gazebo_ros_imu_sensor_world]
(1534543568 309324179) [Err] [Sensor.cc:454] Get noise index not valid
(1534543568 309423449) [Err] [Sensor.cc:454] Get noise index not valid
(1534543568 309449145) [Err] [Sensor.cc:454] Get noise index not valid
(1534543568 309471826) [Err] [Sensor.cc:454] Get noise index not valid
(1534543568 309505286) [Err] [Sensor.cc:454] Get noise index not valid
(1534543568 309554307) [Err] [Sensor.cc:454] Get noise index not valid
(1534543568 398320942) [Dbg] [ServerFixture.cc:209] ServerFixture load in 1 seconds, timeout after 600 seconds
(1534543569 40147125) [Dbg] [ServerFixture.cc:135] ServerFixture::Unload
```

The errors are from failed calls to `sensor->Noise()` in the plugin (https://github.com/ros-simulation/gazebo_ros_pkgs/pull/793/files#diff-43a08bb3c10236c3fea23e88cfcc1d68R81 ) since the noise does not appear to ever be set.

[good_imu_noise_from_cmdline.log](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2298901/good_imu_noise_from_cmdline.log)
[bad_imu_noise_from_test.log](https://github.com/ros-simulation/gazebo_ros_pkgs/files/2298902/bad_imu_noise_from_test.log)

chapulina, 2018-08-18:
I see the errors even when just running Gazebo, both on the terminal and on the log file:

~~~
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.17.0.2
[INFO] [gazebo_ros_node]: ROS was initialized without arguments.
[Err] [Sensor.cc:454] Get noise index not valid
[Err] [Sensor.cc:454] Get noise index not valid
[Err] [Sensor.cc:454] Get noise index not valid
[Err] [Sensor.cc:454] Get noise index not valid
[Err] [Sensor.cc:454] Get noise index not valid
[Err] [Sensor.cc:454] Get noise index not valid
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.17.0.2
~~~

I see a couple differences in the output, I have:

`[INFO] [gazebo_ros_node]: ROS was initialized without arguments.`

And you have:

`(1534544515 487859675) Init world[gazebo_ros_imu_sensor_world]`

dhood, 2018-08-18:
if you always see the error you're maybe testing with a world file that doesn't specify noise, yeah?

chapulina, 2018-08-18:
Yeah I run the test world, just like in the beginning of the issue description, and that doesn't have noise:

`gazebo src/gazebo_ros_pkgs/gazebo_plugins/test/worlds/gazebo_ros_imu_sensor.world --verbose`

dhood, 2018-08-18:
I didn't realise that the test is relying on a version of the world file [copied to the build directory](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/652d57bfcc2c22b17b4531090a63888d1c9915e0/gazebo_plugins/test/CMakeLists.txt#L4). I had modified the world file in the source directory to add noise. Rebuilding didn't trigger a re-copy, so it was still using the world file without noise.

In summary, this was user error. Thanks for taking a look still @chapulina !

closed 2018-08-18 by dhood