ros-simulation/gazebo_ros_pkgs #807: Bad variable name

abhileshborode, 2018-08-26:
The command roslaunch turtlebot_gazebo turtlebot_world.launch  is not working. Its returning

/home/abhilesh/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver: 6: export: :/usr/lib/x86_64-linux-gnu/gazebo-7/plugins: bad variable name
/home/abhilesh/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzclient: 6: export: :/usr/lib/x86_64-linux-gnu/gazebo-7/plugins: bad variable name

[gazebo-2] process has died [pid 20430, exit code 2, cmd /home/abhilesh/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode worlds/empty.world __name:=gazebo __log:=/home/abhilesh/.ros/log/39bd59f0-a985-11e8-adbe-f83441238011/gazebo-2.log].
log file: /home/abhilesh/.ros/log/39bd59f0-a985-11e8-adbe-f83441238011/gazebo-2*.log
[gazebo_gui-3] process has died [pid 20445, exit code 2, cmd /home/abhilesh/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzclient __name:=gazebo_gui __log:=/home/abhilesh/.ros/log/39bd59f0-a985-11e8-adbe-f83441238011/gazebo_gui-3.log].
log file: /home/abhilesh/.ros/log/39bd59f0-a985-11e8-adbe-f83441238011/gazebo_gui-3*.log

Yeah2333, 2018-11-01:
i have this problem .u can edit your bashrc file to add source /usr/share/gazebo/setup.sh .after add to bashrc.you should source .bashrc to reload new bashrc file.but i have a new problem :
/opt/ros/kinetic/lib/gazebo_ros/gzserver: 6: export: :/usr/local/cuda-8.0/lib: bad variable name

abhileshborode, 2018-11-01:
Ya...This  issue was created like 2 months ago, I couldn't find the solution and I was in a hurry that time so I just reinstalled ubuntu along with ros. If you find the solution let us know it could be used by some one else in the future.

Yeah2333, 2018-11-02:
do you have install cuda driver? there are something wrong about bashrc file. if you add export LD_LIBRARY_PATH=/usr/local/cuda-8.0/lib\ ${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}} 
in you bashrc file ,you should delete it and add export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/local/cuda/lib64  to your bashrc file

abhileshborode, 2018-11-03:
Yes I had installed the cuda drivers....Did you solve this issue by just changing the cuda path in bashrc?

Yeah2333, 2018-11-03:
yes.after i change cuda path的source bashrc file.the problem has been solved.