ros-simulation/gazebo_ros_pkgs #812: [gazebo_ros_control] PositionJointInterface and POSITION_PID fail when joint limits are >= pi

azeey, 2018-08-30:
The POSITION_PID controller in gazebo_ros_control uses `angle::shortest_angular_distance_with_limits` which requires that the arguments must lie in [-pi, pi]. So if the joint limits specified in urdf for a revolute joint are exactly [-pi, pi] or larger, the controller fails miserably. 

## Demo
The robot should stay upright when the simulation starts
[image]

## Reproduce
I have created a modified the rrbot example to reproduce the issue.
* (https://github.com/azeey/gazebo_ros_demos/commit/9569b8542dff4cf0eed0af15a6d4150c723b7425) uses position_controllers/JointTrajectoryController
* (https://github.com/azeey/gazebo_ros_demos/commit/b8a37d325b8e45435b7bcdcc245e6f963ffaba34) uses position_controllers/JointPositionController. 

To reproduce, checkout one of the two and run `roslaunch rrbot_gazebo rrbot_world.launch`. To see the effect of the joint limit, edit the `joint_limit` property in `rrbot_description/urdf/rrbot.xacro`.

nyxrobotics, 2023-11-13:
Our projects deal with joints that have a greater range of motion than a full revolution.
For more details on the discussion, please see [here](https://github.com/sbgisen/gazebo_ros_pkgs/issues/2).

We suggest to change here
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/398cde933cba5d72921055d57d4faa47c4445654/gazebo_ros_control/src/default_robot_hw_sim.cpp#L334-L338
to
```cpp
 if(joint_upper_limits_[j] - joint_lower_limits_[j] > 2.0 * M_PI)
  {
    double command = clamp(joint_position_command_[j], joint_lower_limits_[j], joint_upper_limits_[j]);
    error = command - joint_position_[j];
  }
  else
  {
    angles::shortest_angular_distance_with_limits(joint_position_[j],
                                                  joint_position_command_[j],
                                                  joint_lower_limits_[j],
                                                  joint_upper_limits_[j],
                                                  error);
  }
```

The `noetic-devel` branch doesn't seem to be working lately, can I make a patch and create a PR?

mjcarroll, 2023-11-27:
@nyxrobotics were you going to open a PR to address this here?