ros-simulation/gazebo_ros_pkgs #813: Gazebo9 dont provide gazeboConfig.cmake or gazebo-config.cmake?

iDsy, 2018-09-01:
Hi,

I upgrade to Gazebo 9, ROS melodic, ubuntu 18.08. When I compile packages using catkin_make, I get cmake error:

Could not find a package configuration file provided by "gazebo" with any of the following names:

gazeboConfig.cmake
gazebo-config.cmake
Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set "gazebo_DIR" to a directory containing one of the above files. If "gazebo" provides a separate development package or SDK, be sure it has been installed.

I also installed libgazebo9-dev, but I still dont have these .cmake files.

Looking for help! Thanks in advance!

j-rivero, 2018-09-26:
Which version of gazebo are you using?

I found the file in the [online list](https://packages.ubuntu.com/bionic/libgazebo9-dev) for `libgazebo9-dev`on Bionic, version `9.0.0+dfsg5-3ubuntu1`.

littleggghost, 2018-09-26:
I use Ubuntu16.04 + ros Kinetic + gazebo8.

j-rivero, 2018-10-24:
> I use Ubuntu16.04 + ros Kinetic + gazebo8.

I checked [libgazebo8-dev](http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo8/libgazebo8-dev_8.6.0-1~xenial_amd64.deb) package and can see the cmake modules in there.

Please reopen with more details if the error is still valid for your case.

closed 2018-10-24 by j-rivero

Jonnyerdnuss, 2019-04-10:
Hi,

i have the same Problem on my machine.
Ubuntu 18.04.2
ROS Melodic (full installation: http://wiki.ros.org/melodic/Installation/Ubuntu )

If i check 
`gazebo --version` 
i got:
```
Gazebo multi-robot simulator, version 9.8.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

but two times, and if i look for:
`dpkg -L libgazebo9-dev  | grep gazebo-config.cmake`
the output is:
`/usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake`

A short output from the CMakeCache.txt from the build directory:
`
//The directory containing a CMake configuration file for gazebo.
gazebo_DIR:PATH=gazebo_DIR-NOTFOUND
`

I think u have to add a PATH in the CMakeLink.txt or maybe i missed something to add in the bashrc file, but i don't know what.

Thanks!

whwang8, 2022-04-20:
Ubuntu 16.04 Ros Kinetic and Gazebo 7, same issue.

I found the gazebo-config.cmake with 
dpkg -L libgazebo9-dev | grep gazebo-config.cmake

and set gazebo_DIR in CMakeCache.txt to the path found:

/usr/lib/x86_64-linux-gnu/cmake/gazebo/

Solved issue!

whwang8, 2022-04-21:
> Ubuntu 16.04 Ros Kinetic and Gazebo 7, same issue.
> 
> I found the gazebo-config.cmake with dpkg -L libgazebo9-dev | grep gazebo-config.cmake
> 
> and set gazebo_DIR in CMakeCache.txt to the path found:
> 
> /usr/lib/x86_64-linux-gnu/cmake/gazebo/
> 
> Solved issue!

The CMakeCache.txt is found within the build of your catkin_ws