ros-simulation/gazebo_ros_pkgs #832: robot state publisher plugin from "ros-kinetic-gazebo-plugins" sends timestamp 0 with latest debs for ROS kinetic + gazebo9

koenlek, 2018-10-25:
We're using the latest official ROS debs for ROS kinetic on Ubuntu 16.04 from:
- `deb http://packages.ros.org/ros/ubuntu xenial main`

In combination with Gazebo 9.4.1 from 
- `deb http://packages.osrfoundation.org/gazebo/ubuntu xenial main`

Recently, our "robot state publisher" started failing. It didn't show joint states anymore in RVIZ. First I suspected a regression on our side, but after further inspection I think the "bug" is with the "ros-kinetic-gazebo-plugins" package. 

I noticed that the stamp was always 0 when doing `rostopic echo /joint_states`. E.g.:

Example output:
```
header: 
  seq: 2116
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
name: [rear_right_wheel_joint, rear_left_wheel_joint, front_right_wheel_joint, front_left_wheel_joint]
position: [-0.09968497750591343, -0.09197513010067127, 0.2483745999788889, 0.05948497888773652]
velocity: [-0.0020709022831325798, -0.0020246023417283218, -0.002010995487176877, -0.0020249119479750347]
effort: []
```

When I do a git clone of "https://github.com/ros-simulation/gazebo_ros_pkgs" (branch: kinetic-devel) and build `gazebo_plugins` and then run based on my own build of these plugins, the problem disappears. I diffed the code of `gazebo_ros_joint_state_publisher.{cpp,h}` of that code against the source that I get when I run: `apt-get source ros-kinetic-gazebo-plugins`. They are identical. Hence, the problem seems to be that code build in my environment works, but code build in the buildfarm that's used for packaging the debs doesn't work. Pointing to a difference in the compiler, compiler settings, or the build environment?

In late august, ROS released a new version of 'librostime.so' as part of "ros-kinetic-rostime". That seems to have been the cause of many problems for us. We create docker images as build & development environments and we found out that code build in a old version of our docker image didn't run well in our latest version of our docker image. The cause seemed to be a change in librostime.so. The simple solution we found is: rebuild + repackage all software in our latest docker image and run all in that latest docker image. There doesn't seem to be a cleaner solution that's backward or forward compatible. You're either building against the latest ros kinetic debians and your binaries will run fine against those, or your build against the old debs and run fine against those.

I wonder against which version of rostime the current "ros-kinetic-gazebo-plugins"  was build? Because the official deb doesn't work for us, whereas a build from source works fine...

Also, I think that the robot state publisher is probably not the only thing in this repository bothered by this issue. Anything in this repo that uses `ros::Time` could be affected.

koenlek, 2018-10-25:
By the way, this problem should be quite easily reproducable in docker. Consider this dockerfile:

```
FROM osrf/ros:kinetic-desktop
MAINTAINER Koen Lekkerkerker, k.lekkerkerker@robotcaresystems.com

RUN apt-get update && \
    apt-get install -y \
    wget \
    software-properties-common \
 && apt-get clean

RUN echo "deb http://packages.osrfoundation.org/gazebo/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list \
 && wget http://packages.osrfoundation.org/gazebo.key -O - | apt-key add - \
 && apt-get update \
 && apt-get install -y \
    gazebo9 \
    ros-kinetic-gazebo9-ros-pkgs \
 && apt-get clean
```

j-rivero, 2018-11-09:
Thanks for report. The bug is valid, rostime broke the ABI and we need to rebuid packages. I did that but the package builds failed. Working on it now. Let's track it here: https://bitbucket.org/osrf/gazebo/issues/2537/update-gazebo9_ros_pkgs-to-fix-abi

koenlek, 2018-11-09:
Thanks. This is fixed in your new build: https://bitbucket.org/osrf/gazebo/issues/2537/update-gazebo9_ros_pkgs-to-fix-abi

closed 2018-11-09 by koenlek