ros-simulation/gazebo_ros_pkgs #834: Laser plugins time_increment and scan_time zero

reinzor, 2018-11-02:
See:

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_gpu_laser.cpp#L195

and

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_laser.cpp#L190

Some packages require the time_increment to be non-zero, e.g. https://github.com/googlecartographer/cartographer_ros

skohlbr, 2018-11-24:
As noted in #835 it is in fact true that simulated scans generated with currently existing plugins are instantaneous and one has to take care not to publish wrong information to consumers by faking a non-zero scan time, when it is in fact zero.

reinzor, 2018-11-27:
I agree, limited PR to only update the scan_time