ros-simulation/gazebo_ros_pkgs #836: ROS2 - z values accelerating when base_footprint link is added to demo world

mkhansen-intel, 2018-11-06:
Caveat - We're trying to use the ROS2 gazebo_ros_pkgs, which I realize is not fully released yet but @dirk-thomas told us we could try to use it for navigation2 testing and to post issues if we find them.

We ran the demo here:
http://gazebosim.org/tutorials?tut=ros2_installing&branch=ros2#TestingGazeboandROS2integration

Long story short, we discovered that we see strange behavior even in the demo when we add the base_footprint link. 
```
<model name='vehicle'>
      <pose>0 0 0.325 0 -0 0</pose>
      <link name="base_footprint"/>

      <link name='chassis'>
```
We see the odom->chassis seems to be accelerating in the negative Z direction (I'm assuming that's gravity). 

```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws$ ros2 run tf2_ros tf2_echo odom_demo chassis
At time 8.210000000
- Translation: [0.000, 0.000, -329.915]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 9.190000000
- Translation: [0.000, 0.000, -413.465]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 10.180000000
- Translation: [0.000, 0.000, -507.424]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
At time 11.170000000
- Translation: [0.000, 0.000, -610.988]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
```

Is this expected behavior? odom_demo to base_link should be a static transform, right? How does adding a base_footprint change that?

chapulina, 2018-11-06:
> odom_demo to base_link should be a static transform, right?

`odom_demo` is the world frame, so any frames attached to the robot will move w.r.t. odom as the robot moves

I'll try out your specific setup soon.

mkhansen-intel, 2018-11-06:
OK, thanks!

chapulina, 2018-11-06:
tl;dr: The new link doesn't have a collision, so it's falling and dragging the model origin with it.

---

So, what's happening in the example is that the link you added is the first one in the model, so it is the "canonical" link - which means its pose is fixed to the model. But that link doesn't have a collision and is not attached to other links by a joint, so it falls to infinity.

You can check on the Gazebo UI that the `vehicle` model's pose is falling.

 Finally, even though the `chassis` itself is not falling, the TF says `chassis` because that is passed on the configuration: `<robot_base_frame>chassis</robot_base_frame>`. If you omit that tag, the value defaults to `base_footprint`.

mkhansen-intel, 2018-11-06:
OK, so either I remove the base_footprint link, or add a collision for it, right? If the link isn't required, I will remove it. 

Originally I found this problem in the turtlebot3 demo world, which contained the link, so was this something that was needed in an older version of Gazebo and isn't anymore?

chapulina, 2018-11-06:
I haven't looked at the Turtlebot model you're using, but that empty link looks really suspicious, it's not something you'll usually find in SDF. I'd remove it, or maybe add a fixed joint to the chassis? Just adding a collision without connecting it to the rest of the model means the link will just be left behind on the ground when the robot starts moving.

I see the [turtlebot repo's ros2](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/ros2/turtlebot3_gazebo) branch has `COLCON_IGNORE` for `turtlebot_gazebo`, which branch are you working off? I can try taking a look later.

mkhansen-intel, 2018-11-06:
I'm looking at the Turtlebot3 repo here: https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/ros2

I took out the base_footprint link and it seems to work OK now. One other thing I'm seeing however is the turtlebot3 is slowly spinning, not sure why? I was guessing it was due to the friction settings of the wheels but that was just a guess.

chapulina, 2018-11-12:
@mkhansen-intel , I tweaked the turtlebot model a bit, see https://github.com/ROBOTIS-GIT/turtlebot3_simulations/pull/66.

I'm closing this issue, feel free to ticket new ones if you find other problems.

closed 2018-11-12 by chapulina

mkhansen-intel, 2018-11-13:
Thanks @chapulina - we'll give those changes a try